MiczFlor/RPi-Jukebox-RFID #274: Playlist issiues with "No Spotify Version"

toblinga, 2018-10-20:
Hi
i did a new installation from scratch, and now i have found out following issue:

* mpd.conf needs to be updated the playlist-folder to ~/RPi-Jukebox-RFID/playlists
* but the playlists were written as m3u8 instead of m3u therefore the mpd is not able to find the file
* ==> Oct 20 22:33 : errno: Failed to open "/home/pi/RPi-Jukebox-RFID/playlists/Benjamin BlÃ¼mchen.m3u": No such file or directory
* i changed the the playlistname to m3u, => playing is working, but in the web ui the playlist is empty 

[image]

woll-knaeuel, 2018-10-21:
Hello
I can confirm the same problem with the "Spotify Version" 1.1.7.
New installation on an empty SD card.
Which files needs to be edited as a work around till this issue has been fixed with an updated version?
As described by toblinga either mpd is working or the web ui.

toblinga, 2018-10-21:
Hi,
as work around you can change playlist path in the mod.conf to
~/rpi-jukebox-rfid/playlists and rename the playlists in this folder from
*.m3u8 to m3u

If you don't find any file in the playlist start some music over the web UI

I do not know if the intention is to use m3u8 for mpd as well or to switch
between m3u and m3u8, depending of the player-software.

Cheers

MiczFlor, 2018-10-21:
Hi @toblinga @woll-knaeuel 

the `m3u8` extension should not do any harm. the `8` signifies that the playlist contains UTF-8 characters.
However: the path to the playlistfolder could and should be a problem. Try to run this command:
~~~
sudo cat /etc/mpd.conf | grep playlist_directory
~~~
if it returns `playlist_directory		"/home/pi/RPi-Jukebox-RFID/playlists"` you should be fine. If it returns `playlist_directory		"/tmp" you are in trouble, because then the Phoniebox generates playlists in a different directory than the one where mpd is looking for playout. This one line should fix it:
~~~
sudo sed -i 's|/tmp|/home/pi/RPi-Jukebox-RFID/playlists|' /etc/mpd.conf
~~~
Then try again the above command and it should return the (new) correct location for playlists

Then reboot.

Please share here if this helps and what else it breaks. Possibly try this again with the `m3u8` playlist to see if this really is a problem.

Apologies, the spotify update has done some changes that I haven't caught in the release.

toblinga, 2018-10-21:
hi, thanks for the fast response:
i changed already the mpd.conf (should be changed also in the mpd.conf.sample (i'll make a pull request), and after changing back the m3u8 -> m3u in rfid_trigger_play.sh it is working fine again.

it seams mpd does not allow m3u8 files...

Thanks you

MiczFlor, 2018-10-22:
@toblinga @woll-knaeuel  
In the `develop` branch, I added the option to change the file ending for playlists. See here for details:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/0f31aeb0f5104cb626b0bf4e2795d1e8169b82cd
By default `m3u` is set. 
This will be merged to master once somebody confirms it works for them, too. Thanks

toblinga, 2018-10-22:
@MiczFlor 
with changing back to m3u the box is playing again...
but the playlist on the web UI the tracknames are not shown. 
[image]
but i don't know if this bug has something to do with the playlist-ending or is something different...

princemaxwell, 2018-10-23:
This could have to do with MPD himself or missing ID3Tags but I think it's MPD. To get the content of playlist I integrated a command which read the playlist info. It could be that this is only working with mopidy. It could be that MPD can't interpret this command. I will investigate this. If that's the fault, I change the playlist back to filenames for the MPD version.

toblinga, 2018-10-23:
@princemaxwell this files should have I3DTags, therefore i assume it has something to do with MPD and the playlist command.

princemaxwell, 2018-10-23:
@MiczFlor 
After testing with two different SD Cards (spotify & classic version) i found a solution playlist file extension handling. For both version we can use M3U, so we have M3U as standard. We only have to add some configurations to mopidy.conf.
Shall i pull request this or do you want to keep M3U and M3U8 both?

@toblinga 
i tested it in my pure MPD version. i have no problem with the playlist. My informations are shown exactly.

[image]

Your screenshot shows, that song time is loaded. So the files are read. There could be a problem with your title and artist tags...please check this...

toblinga, 2018-10-23:
intresting, which version are you using?

princemaxwell, 2018-10-24:
The screenshot shows the actual master. I did a fresh install with one line script...

toblinga, 2018-10-24:
i agree, for I3dtags it works, but for files without i3dtags the playlist shows no title, file name... just the duration.
it would be good to add a check, if the file has i3dtags incluced, otherwise show just the filename

princemaxwell, 2018-10-25:
@toblinga
Try the 1.1.8-beta I pull requested.
You have to wait until it's merged to develop branch.

I haven't tested your issue, but the file handling is better in the new version.

MiczFlor, 2018-10-30:
Hi @toblinga 
the 1.1.8-beta is now the master branch. Possibly have a test and see if it works. I am trying to close some threads :)

toblinga, 2018-10-31:
Hi
Thanks I'm on holiday for the next days, so i can test it next week.
Cheers
Armin

MiczFlor, 2018-11-21:
The code has moved on. I am closing this, assuming the later code solved it. :) All the best, micz

closed 2018-11-21 by MiczFlor