MiczFlor/RPi-Jukebox-RFID #276: RFID-swipes are recognized, but audio does not start playing anymore

TrustCompact, 2018-10-20:
Hey,

Thumbs up for this great project. 
I had everything running this afternoon. As I started to upload more and more music, I created a bunch of subfolders to organize things. From this moment on, no audio has been played anymore. 
If I swipe an RFID-card, the code is recognized and the correct folder is selected in the webbrowser (I can see this because cover image changes properly). But the file is not played. Also a manual click on the play button does not start playing.
I changed nothing except the subfolders and already tried to update owner and permission on the folder without success. 

OS is 2018-10-09-raspbian-stretch-lite
I ran the one-script-installer;
All settings are default so far.
Soudcard is a Creative X7 USB

As mentioned - everything worked fine so far, until modifying folder-structure.
(I re-assigned cardIds to new folders, of course)

If you need further information please let me know!

Any help would be appreciated!

pgx3d, 2018-10-21:
I have updated to the newest version. Now I‘m not able to start Play-off the files. Not over the web gui nor with the cards.

I also made a fresh install on a clean sd card. 
Reversing to 1.16 didn‘t work either.

BR
Manuel

ekorn, 2018-10-21:
same for me. The  ./scripts/rfid_trigger_play.sh create in my /tmp directory .m3u8  files if I rename it to.m3u the work again.

princemaxwell, 2018-10-21:
As i build the Spotify version, I forgot to configure the MPD.conf for the no Spotify version.

There were some changes necessary for mopidy in the Triggerfile. M3U8 has to set as default playlist scheme and the path has to be changed relating to the configuration.

I will work on this. That's my fault.

TrustCompact, 2018-10-21:
> Please share here if this helps and what else it breaks. Apologies, the spotify update has done some changes that I haven't caught in the release.

Thanks for your quick response.
I tried the way you described - my path was set to `playlist_directory	"/tmp" and it could be halfly fixed with your solution.

Using WebApp and/or Putty console now, sometimes card swipes result in 
- nothing except reading the code
- switching the cover in the web frontend and do not play
- switching the cover in the web frontend and play (with big lag)

all in all something seems to be happened which made playing a little buggy!

The title information while playing (eg playtime/position ...) still is gone.

princemaxwell, 2018-10-21:
Have you installed the Spotify version?
Only there is a little lag when playing local music.
If you use the original MPD version without Spotify, there is no lag.

Title information should work, are you sure your MP3s have ID3 tags?

ekorn, 2018-10-21:
in my mpd conf was the /tmp directory setting. But in /home/pi/RPi-Jukebox-RFID/settings/Playlists_Folders_Path was also /tmp. I changed it to /home/pi/RPi-Jukebox-RFID/playlists but when I sent  ./scripts/rfid_trigger_play.sh -i=XXX it still get:
mpd error: No such playlist
via web interface it does not work ether

MiczFlor, 2018-10-22:
@princemaxwell @ekorn 
In the `develop` branch, I added the option to change the file ending for playlists. See here for details:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/0f31aeb0f5104cb626b0bf4e2795d1e8169b82cd
By default `m3u` is set. 
@princemaxwell 
could you work with the latest `develop` version and use this feature, please? Who knows what the file end option will be useful for in the future :)

roland17387, 2018-10-22:
my path was also set to "/tmp" 
as discribed it changed to "/home/pi/RPi-Jukebox-RFID/playlists"
but music still does not play over webapp or card swipe

MiczFlor, 2018-10-22:
@roland17387 

Go to the Phoniebox dir and update the code with
~~~
git pull
~~~
Then try to run this command:
~~~
sudo cat /etc/mpd.conf | grep playlist_directory
~~~
if it returns `playlist_directory		"/home/pi/RPi-Jukebox-RFID/playlists"` you should be fine. If it returns `playlist_directory		"/tmp" you are in trouble, because then the Phoniebox generates playlists in a different directory than the one where mpd is looking for playout. This one line should fix it:
~~~
sudo sed -i 's|/tmp|/home/pi/RPi-Jukebox-RFID/playlists|' /etc/mpd.conf
~~~
Then try again the above command and it should return the (new) correct location for playlists

Then reboot.

Does that fix it?

roland17387, 2018-10-22:
yes, everything works now, thank you,
one folder still did not play after the update, but when i deleted the files and uploaded them again in this folder, everything worked

ekorn, 2018-10-22:
For me it worked partly. Now m3u files are created in the playlist folder and if I choose a card connected with a single folder the playlist is filled with the path to files. But if I choose a card that is connected with a folder containing multiple sub folders a empty 0 Byte m3u file will be created. If start the folder from the webinterface it works fine.  But the playlist file is owned by root and has the filename:
<FolderName> %RCRSV%.m3u
if I rename the file to <FolderName>.m3u the card works fine. Next time start the folder from the webinterface leads to another  %RCRSV% file

princemaxwell, 2018-10-31:
### EVERYONE WITH PROBLEMS STARTING LOCAL PLAYLISTS HAS TO RUN THIS COMMAND ONCE:

Depending of the location of audiofolder (this is for default location):
`sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/shared/audiofolders/`

### If you have +Spotify Edition (>1.1.8-beta) you have to SCAN MUSIC LIBRARY
-> Renew your database under Folders & Files.

And please upload this file to fix the upload in the WebUI or wait until @MiczFlor has merged it!
https://github.com/princemaxwell/RPi-Jukebox-RFID/commit/500922f1f768d21e334844ab0f2a047a8cddba90

closed 2018-11-21 by MiczFlor