MiczFlor/RPi-Jukebox-RFID #277: Upload über Web-App

roland17387, 2018-10-21:
Hallo, 
habe jetzt meine zweite Phoniebox gebaut. Bei der zweiten funktioniert aber der Dateiupload über Webapp nicht. Die erste Phoniebox hat aber auch noch eine alte Software (jede Datei muss einzeln zum Upload ausgewählt werden, nicht mehrere Dateien aus einem Ordner auf einmal auswählbar)
Die Einstellungen sind wie im Tip "file_uploads = On" usw.
Über Netzwerk (Samba) kann ich Dateien hinzufügen.
Wenn ich Dateien über Webapp hochladen will und einen Namen für einen neuen Ordner eingebe, dauert es einen Moment, als würden die Dateien übertragen, am Ende sind aber weder die Audiodateien noch ein neuer Ordner da.
Einen neuen Ordner kann ich aber erstellen. Aber auch in einen bereits erstellten Ordner kann ich keine Dateien hochladen.
Die Dateien, die ich über Samba hochlade, kann ich leider nicht über die Phoniebox wiedergeben, weder über Webapp noch über eine zugewiesene RFID-Karte.

kaoz3000, 2018-10-21:
Vielleicht stimmen die Zugriffsrechte nicht?
Versuche es mal mit:
```
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
```

roland17387, 2018-10-21:
Hat mit dem Tip halbwegs funktioniert, manchmal wurde aber nur der Ordner erstellt, aber datei war nicht enthalten, manchmal hat weiterhin gar nichts funktioniert, manche dateien konnte ich dann nicht mehr löschen. habe dann nochmal die one line installation laufen lassen, jetzt kann ich dateien hochladen. nur abspielen funktioniert nicht, könnte aber auch ein anderes problem sein (siehe #276 )

closed 2018-10-21 by roland17387