MiczFlor/RPi-Jukebox-RFID #280: no covers

e-p-s, 2018-10-22:
I have the covers both in the mp3's and again as folder.jpg in the respective folders.

unfortunately i still don't get a cover. does anyone have an idea?

MiczFlor, 2018-10-22:
You remembered the logic correctly but not the file name (from the manual): 

Show covers in web app

If your audio folder contains a file called cover.jpg (lowercase!) it will be displayed in the web app above the player controls.

—-

Reading from the mp3 file is theoretically possible but hasn‘t been written - also because it‘s anothe CPU load that could be avoided.

e-p-s, 2018-10-23:
thanks. okay. 
i just named it folder.jpg ex works, because it's the usual spelling in windows.

wouldn't it be possible to query whether cover.jpg or folder.jpg exists and then select it?

furthermore, if there is no cover, there would be a nice way to hide the cover field so that the media control moves up again.

Background of this, I have installed a Raspberry with 3.5" display in my small children's box.
If there's a cover it's nice, if not it would be nice if you could get it directly to the controls.

Oh yes a smaller title line (optional) would also be great, because a lot of space is lost at the moment.

Maybe you can create a mini theme.

s-martin, 2023-12-28:
Version 2.x uses cover.jpg and version 3.x supports only covers integrated in the MP3 file.

Closing this now.

closed 2023-12-28 by s-martin