MiczFlor/RPi-Jukebox-RFID #283: Integration of Smart Automatic Cover Art Downloader

hankman13, 2018-10-22:
This would be a nice feature to download the cover art.
[https://github.com/desbma/sacad](https://github.com/desbma/sacad) requires python 3, so it has to be compiled on your own.
I couldn't test it these days, but it doesn't look so hard to get it running.

hankman13, 2018-10-22:
First feedback, I just tried the windows binaries for the media library of some test folders. There are some false recognized covers. Maybe it's possible to imrove the detection rate, anyway I just got very fast some cover images.

gizmo21, 2018-10-30:
I added the package "abcde" to autorip the old kiddies CDs to the phoniebox and in this package a bin called https://github.com/sahib/glyr is used to get coverart automagically while ripping/encoding the CD, which here and then fails for german kiddie-CDs.

So if your folders are named properly to detect artist and albumname or your files are metatagged well, your option would be nice to get the rest of the not autoripped covers fast.

closed 2019-06-21 by MiczFlor