MiczFlor/RPi-Jukebox-RFID #284: Bluetooth Audio

Ardios, 2018-10-23:
Hallo zusammen,

ich möchte euch hier kurz eine Anleitung schreiben, für alle die gerne ihre Phoniebox mit Bluetooth Lautsprechern, Kopfhörern etc. verbinden möchten.

Es wäre von Vorteil, wenn ihr eure Phoniebox schon vorinstalliert habt (Ich nutze bei mir die Standarteinstellungen in allen Punkten) da auch die mpd.conf angepasst werden muss.

Meine Basis ist ein Raspberry Pi Zero W, das aktuelle Raspbian Image "2018-10-09-raspbian-stretch-lite" und Phoniebox 1.1.7 ohne Spotify (Die Version mit Spotify habe ich bisher noch nicht getestet) sonst wurde nichts installiert.

Als erstes führen wir ein Update durch:
`sudo apt update && sudo apt upgrade`

Als nächstes benötigen wir bluealsa (In der Desktop Version ist das Paket bereits vorinstalliert):
`sudo apt install bluealsa`

Nun bitte die Phoniebox neustarten, dies ist notwendig! Ich nutze einfach:
`sudo reboot`

Jetzt können wir unseren Bluetooth Lautsprecher/Köpfhörer einschalten und starten danach bluetoothctl:
`sudo bluetoothctl`

Dann geben wir folgende Befehle nacheinander ein:
`power on`
`agent on`
`scan on`

Nun sucht eure Phoniebox nach Bluetooth Geräten in der nähe, bei mir sieht das z.B. so aus:
`pi@phoniebox:~ $ sudo bluetoothctl`
`[NEW] Controller B8:27:EB:26:53:17 phoniebox [default]`
`[bluetooth]# power on`
`Changing power on succeeded`
`[bluetooth]# agent on`
`Agent registered`
`[bluetooth]# scan on`
`Discovery started`
`[CHG] Controller B8:27:EB:26:53:17 Discovering: yes`
`[NEW] Device FC:58:FA:DD:AA:D2 MD 43438`

In meinem Fall ist dies mein Lautsprecher (Kopiert bzw. notiert euch die Mac Adresse eures Lautsprechers, diese benötigen wir später noch):
`[NEW] Device FC:58:FA:DD:AA:D2 MD 43438`

Als nächstes folgt das Pairing (Anstelle von FC:58:FA:DD:AA:D2 müsst ihr die MAC Adresse eures Bluetooth Lautsprechers angeben):
`pair FC:58:FA:DD:AA:D2`

Danach müssen wir dem Bluetooth Gerät noch vertrauen also:
`trust FC:58:FA:DD:AA:D2`

Jetzt können wir uns mit unserem Lautsprecher verbinden:
`connect FC:58:FA:DD:AA:D2`

Nun beenden wir noch den Discovery Modus und können bluetoothctl danach verlassen:
`scan off`
`exit`

Nun legen wir eine neue Datei an:
`sudo nano /etc/asound.conf`

Mit folgendem Inhalt (Bitte Mac Adresse anpassen, pcm.btspeaker kann bei Bedarf auch angepasst werden z.B pcm.sonosplay genau wie die Beschreibung):

pcm.btspeaker {

    type plug

    slave.pcm {

        type bluealsa

        interface "hci0"

        device "FC:58:FA:DD:AA:D2"

        profile "a2dp"

        delay -20000

    }

    hint {

        show on

        description "Medion Partybox"

    }

}

Wir können nun mit folgendem Befehl testen ob die Soundausgabe funktioniert (Falls ihr pcm.btspeaker umbenannt habt, müsst ihr btspeaker durch euren Namen ersetzen):
`aplay -D btspeaker /usr/share/sounds/alsa/Front_Center.wav`

Es ist fast geschafft, nun müssen wir lediglich noch den Audio Output in der mpd.conf editieren:
`sudo nano /etc/mpd.conf`

Hier suchen wir nach audio_output und passen folgende Zeilen an (Auch hier bitte wieder eure Mac Adresse eintragen):

        type            "alsa"
        name            "My ALSA Device"
        device          "bluealsa:HCI=hci0,DEV=FC:58:FA:DD:AA:D2,PROFILE=a2dp"        # optional
        mixer_type      "software"      # optional

Nun noch speichern und das war's. 

Hoffe dem ein oder anderen damit weiterhelfen zu können. Ich werde nachher noch eine kurze, nicht so ausführliche, Zusammenfassung schreiben.

MiczFlor, 2018-10-24:
Hi @Ardios 
this is great. Is there any chance you could write a sketchy version in English for me to edit and add to the manual?

princemaxwell, 2018-10-27:
@Ardios 
If i had installed it...
Is there a way to easily switch from to bluetooth audio without having to reboot?
My thought are going to a swipe action. Swiping a card switches audio output to my bluetooth device?!?!

Ardios, 2018-11-03:
@princemaxwell 

ich habe mir mehrere mpd.conf erstellt und diese unter /home/pi/scripts/ gespeichert z.B mpd.conf_hdmi, mpd.conf_bt etc.

Beim swipe lasse ich dann folgenden Befehl ausführen:

`sudo cp /home/pi/scripts/mpd.conf_bt /etc/mpd.conf && sudo service mpd restart `  

Vielleicht nicht die beste Lösung aber dadurch erspare ich mir einen kompletten reboot.

kle99, 2018-12-02:
you can use different audio_outputs in mpd.conf:
`audio_output {
        type            "alsa"
        name            "My ALSA Device"    # change if you like
        device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
        mixer_type      "software"
#       mixer_device    "default"       # optional
#       mixer_control   "PCM"           # optional
#       mixer_index     "0"             # optional
}
audio_output {
        type            "alsa"
        name            "Medion Partybox"    # change if you like
        device          "btspeaker"
        mixer_type      "software"
        auto_resample   "no"
        auto_format     "no"
        enabled         "no"
}`
1.) A word on mpd audio_output device
@Ardios  defines the alsa pcm device twice. First in `/etc/asound.conf` and secondly in `/etc/mpd.conf`. That is not neccessary.
2.) A word on `enabled         "no"` 
The bluetooth connect doesn't work always, if the mpd audio_output is enabled. 
Enable with `mpc enable <output_no>`. <output_no> is the decimal number of the output shown with `mpc outputs`

ShaXiaomi, 2018-12-07:
Hello,

I was able to set up bluetooth audio for the Spotify version!
To get my speaker to work I had to change some things.
Had to remove "interface" and "delay" in the asound.conf.
/etc/asound.conf:
`
pcm.btspeaker {
type plug
slave.pcm {
type bluealsa
device “5C:93:A2:8B:EE:81”
profile “a2dp”
}
hint {
show on
description “SRS-X77”
}
}`

mopidy.conf:
`[audio]
output = alsasink device=btspeaker
mixer_volume = 30`

Does anyone know how I can add a "fallback to internal speaker" for Mopidy?
As suggest above I would now try just to change the mopidy.conf. But I think this should be possible without any scripts or manual editing.
The best would be, if Mopidy just realizes that the BT speaker is not connected and falls back to a "default" speaker setup.

Any ideas?
Thanks 
Daniel

puked79, 2018-12-09:
Hallo und danke.

@Ardios  Ich habe eine Frage: Wie führe ich denn einen Befehl per Swipe aus? Blöde Frage vermutlich, aber ich stehe da etwas auf dem Schlauch.

romanbalsiger, 2018-12-22:
Hi @Ardios 
i can't get it to work. i copied everything from your example (except the MAC Adress) and i get the test sound to play, but no output from mpc.
what is your audio_output exactly? What's the content of your settings/Audio_iFace_Name?

freddymcfreddfredd, 2018-12-29:
Hey,

I have the same problem as @romanbalsiger . I followed all the instructions and I am able to play the sample sound. 

When I enter
`mpc outputs`
I get the following list
`Output 1 (Mute) is disabled`

Enabling Output 1 ends the wired speaker's playback, but does not start any playback in my bluetooth headphones.  Interestingly enough, mpc tells me, that the playback is indeed running, but I cannot hear anything. 

`Queen - 20th Century Fox Fanfare
[playing] #1/22   0:11/0:25 (44%)
volume: 30%   repeat: off   random: off   single: off   consume: off`

Disabling Output 1 makes the playback come back to the internal speakers. 

Also, I'm puzzled by the word `(Mute)` in the mpc listing. Alsamixer tells me that there is nothing muted in my system. 

I have a USB-soundcard connected to my Pi. Not sure if that could have anything to do with that. (?)
Thanks in Advance!

Ghostsl84, 2018-12-30:
Viielen Dank für die Klasse Anleitung.
Ich habe alles dannach gemacht und bekommne auch durch den Befehl eine Tonausgabe und wenn ich auf der Desktop Oberfläche bei Raspbian bin und Musik spiele bekomme ich ebenfalls Ton aber wenn ich die Musik über Phoniebox abspiele bekomme ich keinen Ton über Bluetooth

habe ich etwas vergessen einzustellen?

Besten Dank

comdoxx, 2019-01-08:
Hi,
this sounds great so far. But as @ShaXiaomi already mentioned, I'd also like to switch to and from bluetooth automatically. So the trigger for that has to be a kind of surveillance of the network state of the device. Haven't figured this out, how to achieve it, maybe anyone has done that before?

Ghostsl84, 2019-01-31:
Hallo Ich bins nochmal,

ich habe jetzt wirklich Tagelang probiert und es genau so wie oben beschrieben eingestellt.
Leider bekomme ich einfach keine Tonausgabe über Bluetooth hin.
mit dem Befehlt 
aplay -D btspeaker /usr/share/sounds/alsa/Front_Center.wav 
funktioniert Bluetooth super,
doch leider bekomme ich keine eine Tonausgabe über Bluetooth wenn die Hörpsiele laufen .
Bluetooth ist beim Raspi immer eingeschaltet.
Schließe ich aber ein Kabel an, habe ich sofort Ton und alles funktioniert.
Vielleicht weiß jemand mehr.

Besten Dank
Ghostsl84

Ardios, 2019-01-31:
@Ghostsl84 

Welche Version nutzt du? Die Classic oder Spotify?

Was hast du denn in der mpd.conf unter audio_outputs eingetragen? (Falls du die Spotify Version nutzt, wäre auch gut zu wissen was du in der mopidy.conf eingetragen hast)

Wenn es mit aplay läuft, ist es die halbe Miete ;)

Ghostsl84, 2019-01-31:
Guten Abend,

ich besitze folgende Version

OS Distribution
Raspbian
Description
Raspbian GNU/Linux 9.6 (stretch)
Release
9.6
Codename
stretch
Phoniebox Setup
Version
1.1.9-rc3
Edition
Plus edition (feat. Spotify integration)
Mopidy Server Status
active (running) since Thu 2019-01-31 13:29:10 CET; 6h ago

unter mpd.conf - audio_outputs sieht es so aus

# Audio Output ################################################################
#
# MPD supports various audio output types, as well as playing through multiple
# audio outputs at the same time, through multiple audio_output settings
# blocks. Setting this block is optional, though the server will only attempt
# autodetection for one sound card.
#
# An example of an ALSA output:
#
audio_output {
        type            "alsa"
        name            "SoundCore 2"
#       device          "bluealsa:HCI=hci0,DEV=08:EB:ED:E1:B9:28,PROFILE=a2dp"      # optional
#       mixer_type      "software"      # optional
#       mixer_device    "default"       # optional
#       mixer_control   "PCM"           # optional
#       mixer_index     "0"             # optional
}
#
# An example of an OSS output:

Wenn ich die mopidy.conf mit dem Befehl ~/.config/mopidy/mopidy.conf öffnen möchte, steht dort "Keine Berechtigung"
an dieser Datei habe ich bis jetzt auch nichts geändert.

Gruß
Ghostsl84

Ghostsl84, 2019-01-31:
Guten Abend,

habe das System jetzt neu aufgesetzt OHNE Spotify.
Wenn ich jetzt Musik über den Raspi direkt abspiele (Desktop Oberfläche) auf Rasbian Stretch dann bekomme ich die Musik über den Bluetooth Lautsprecher ausgegeben.
Benutze ich aber eine RFID-Karte für Phoniebox dann bekomme ich keine Tonausgabe über Bluetooth und auch keine mehr über 3,5 mm Klinkenstecker
Ich verzweifle noch :-)

Nächtliche Grüße
Ghostsl84

Ghostsl84, 2019-02-01:
Guten Morgen,

das funktioniert leider nicht. Wenn ich die MP3s über das Webinterface (Phoniebox Oberfläche) abspiele kommt kein Ton raus.
Spiele ich es aber über Rasbian (Desktop Oberfläche) funktioniert alles super über Bluetooth
Muss ich unter Raspi-config noch etwas ändern?
oder unter Alsamixer?
Bluetooth Verbindung zu Rasbian Desktop Oberfläche ist beim Neustart / Reboot direkt da

Ardios, 2019-02-04:
@Ghostsl84 

Bitte entschuldige die späte Rückmeldung...
So habe nun auch nochmal frisch die aktuellste Raspbian Stretch Lite installiert und ex extra mit einem RPi 3b+ getestet.

Habe mit den Updates angefangen, dann Phoniebox installiert...reboot...dann habe ich bluealsa installiert, `sudo bluetoothctl`, Pairing, trust, connect (Hast du ja auch alles soweit gemacht.)

Die asound habe ich dieses mal weggelassen...also direkt `sudo /etc/mpd.conf` und meinen Audio Output angepasst:

[image]

speichern, danach noch schnell `sudo service mpd restart` ...danach lief es direkt über Webinterface...RFID habe ich jetzt noch nicht getestet, wird aber auch laufen.

Ich kann mir nicht vorstellen das es an der Desktop Version liegt aber irgendwo ist der Wurm drin

Ghostsl84, 2019-02-04:
Guten Abend,

vielen Dank für deine Hilfe jedoch komme ich immer noch nicht weiter. habe jetzt mal systemctl status bluetooth gemacht und das kam raus:

pi@raspberrypi:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset
   Active: active (running) since Mon 2019-02-04 19:51:33 CET; 10min ago
     Docs: man:bluetoothd(8)
 Main PID: 767 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─767 /usr/lib/bluetooth/bluetoothd

Feb 04 19:51:33 raspberrypi bluetoothd[767]: Bluetooth management interface 1.14
Feb 04 19:51:33 raspberrypi bluetoothd[767]: Failed to obtain handles for "Servi
Feb 04 19:51:33 raspberrypi bluetoothd[767]: Sap driver initialization failed.
Feb 04 19:51:33 raspberrypi bluetoothd[767]: sap-server: Operation not permitted
Feb 04 19:51:33 raspberrypi bluetoothd[767]: Endpoint registered: sender=:1.6 pa
Feb 04 19:51:34 raspberrypi bluetoothd[767]: Failed to set privacy: Rejected (0x
Feb 04 19:51:48 raspberrypi bluetoothd[767]: Endpoint registered: sender=:1.6 pa
Feb 04 19:56:10 raspberrypi bluetoothd[767]: Unable to get io data for Hands-Fre
Feb 04 19:56:40 raspberrypi bluetoothd[767]: Unable to get io data for Hands-Fre
Feb 04 19:58:06 raspberrypi bluetoothd[767]: Unable to get io data for Hands-Fre
lines 1-19/19 (END)

könnte es sein, das es etwas damit zu tun hat?
Ich bekomme über das Webinterface keinen einizigen Ton raus aber über die Desktopoberfläche des Rasbian Systems jedoch klappt alles wunderbar.

Besten Dank
Ghosty84

Ghostsl84, 2019-04-08:
Einen wunderschönen guten Abend,

ich wollte mal nachfragen, ob es eventuell jetzt machbar ist, über eine Bluetooth Box den Ton auszugeben?
Ich habe mittlerweile 12 verschiedene Bluetooth Boxen ausprobiert und alle führen zum gleichen Ergebnis - Kein Sound nur über 3,5 mm Klinke.

Gruß
Ghosty84

Ardios, 2019-04-11:
@Ghostsl84 

Sorry für die späte Antwort. Ich erstelle dir nachher ein Tutorial inkl. Bilder....ich habe keinerlei Probleme mit der Anleitung von oben.

Nutzt du die Classic oder Spotify Edition?

Viele Grüße
Ardios

Ghostsl84, 2019-04-14:
Guten Abend,

ichz nutze die Classic version 
Bitte Entschuldige die späte Antwort :-)

Vielen Dank
Ghostsl84

closed 2019-06-21 by MiczFlor

prebm, 2019-11-13:
@MiczFlor 
 For the Buster installation I had problems getting audio output and the following error:
```
aplay: main:828: audio open error: Invalid argument
```

I was able to resolve the issue by changing the settings suggested in this issue https://github.com/Arkq/bluez-alsa/issues/234. The updated files look like this:

`asound.conf`

```conf
pcm.btspeaker {
    type plug
    slave.pcm {
        type bluealsa
        service "org.bluealsa"
        device "2C:41:A1:2B:20:79"
        profile "a2dp"
        delay -20000
    }
    hint {
        show on
        description "My BT Device"
    }
}
```

`mpd.conf`

```conf
        type            "alsa"
        name            "My ALSA Device"
        device          "bluealsa:SRV=org.bluealsa,DEV=FC:58:FA:DD:AA:D2,PROFILE=a2dp"        # optional
        mixer_type      "software"      # optional
```

I don't know if this is an Buster issue because it is my first install, but I can confirm now that it works with Buster on a Raspberry Zero with a Bose BT device.

Tolbag, 2020-05-01:
Hi,
I'm on buster and installed the current Developer install to use spotify.

I dont get phoniebox to use the bluetooth box, which is connected. The test sound is fine, but the music comes out of the connected boxes.

Can somebody help?

my asound.conf:
```

pcm.btspeaker {
    type plug
    slave.pcm {
        type bluealsa
        service "org.bluealsa"
        device "FC:58:FA:34:78:B8"
        profile "a2dp"
        delay -20000
    }
    hint {
        show on
        description "Medion Partybox"
    }
}
```

and my mpd.conf
```

 An example of an ALSA output:
#
audio_output {
        type            "alsa"
        name            "My ALSA Device"
        device          "bluealsa:SRV=org.bluealsa,DEV=FC:58:FA:34:78:B8,PROFILE=a2dp"  # optional
        mixer_type      "software"      # optional
#       mixer_device    "default"       # optional
#       mixer_control   "PCM"           # optional
#       mixer_index     "0"             # optional

```

kolmberger, 2020-11-08:
Hey community, 
thank you for all your sharing and help. I hope, my question is placed correctly here, if not - please let me know, I am not a Github-Pro.

On my Phoniebox I'd like to switch over between the hifiberryminiamp-audio-output to the speakers and the wired, plugged-in headphones. The idea is, to do that, when a card is swiped. 

I am using the hifiberryminiamp, which works great standalone. For enhanced use of my kids, e.g. on a longer car-drive I'd like to have them using their headphones with the box.

How do i approach the toggling between output over hifiberryminiamp-->speakers and 3.5mm audio-jack-->headphones?
What I tried so far:

in file /boot/config.txt
```
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtoverlay=hifiberry-dac
``` 
--> that only makes the audio-plug available in raspi-config, I found out

in file /etc/mpd.conf
```
audio_output {
        type            "alsa"
        name            "hifiberryminiamp"
#       device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
        mixer_control   "Master"                # optional
#       mixer_index     "0"             # optional
}
#audio_output {
        type            "alsa"
        name            "Headphones"    # change if you like
        device          "Headphones"
        mixer_type      "software"
        auto_resample   "no"
        auto_format     "no"
        enabled         "no"
}
```

commands that I tried:
`sudo mpc enable 2 && sudo mpc disable 1 && sudo mpc play`
result is no audio at all, see
```
Output 1 (hifiberryminiamp) is enabled
Output 2 (Kopfhoerer) is enabled
Output 1 (hifiberryminiamp) is disabled
Output 2 (Kopfhoerer) is enabled
pi@phoniebox1:~ $ mpc play
ROCK ANTENNE: INXS - Mystify
[paused]  #1/1   0:00/0:00 (0%)
volume:  1%   repeat: off   random: off   single: off   consume: off
ERROR: Failed to open "Kopfhoerer" (alsa); Failed to open ALSA device "headphone": No such file or directory
```

`sudo mpc enable 1 && sudo mpc disable 2 && sudo mpc play`
result is audio over headphones:
```
Output 1 (hifiberryminiamp) is enabled
Output 2 (Kopfhoerer) is enabled
Output 1 (hifiberryminiamp) is enabled
Output 2 (Kopfhoerer) is disabled
ROCK ANTENNE: INXS - Mystify
[playing] #1/1  18:20/0:00 (0%)
volume:100%   repeat: off   random: off   single: off   consume: off
```

How can I get it working?
Thanks for your help!

groows, 2021-01-10:
> Hello,
>
> I was able to set up bluetooth audio for the Spotify version!
> To get my speaker to work I had to change some things.
> Had to remove "interface" and "delay" in the asound.conf. [...]

Hi everybody and thanks for all those informations and help.
Removing of "interface" and "delay" from the asound.conf also solved my problems.

s-martin, 2021-01-10:
Hey, you might want to check #1262

golere, 2022-01-09:
Probably someone can help me here, I can't find the error in my configuration. I'm on a RPi zero 2 W with Buster.

I followed this Tutorial https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Bluetooth and when I use
`aplay -D btspeaker /usr/share/sounds/alsa/Front_Center.wav`
everything is fine and I hear the sound through the BT-Headphones
`Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono`

But using 
`aplay /usr/share/sounds/alsa/Front_Center.wav`
gives me:
`aplay /usr/share/sounds/alsa/Front_Center.wav
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM default
aplay: main:828: audio open error: No such file or directory`

` cat /proc/asound/modules`
gives me nothing because no soundcard is connected

Anybody can help me what I'm missing?

PittBie, 2024-11-20:
Hi, sadly this guide seems a bit outdated.
When I make a clean install I am suggested to use the legacy version for Raspberry Pi Zero WH, which is based on Bullseye. But for Bullseye `Bluealsa is not available`. I found this in a forum: https://forums.raspberrypi.com/viewtopic.php?t=344797

How can I use a bluetooth speaker if Bluealsa is not available?
Trying to install it with `sudo apt install bluealsa` returns me:
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package bluealsa is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'bluealsa' has no installation candidate
```

Any ideas? Much appreciated.

s-martin, 2024-11-20:
For the Phoniebox Version 2 this should be the current guide: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Connecting_Bluetooth_device_to_Phoniebox

For Version 3:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/audio.md#bluetooth