MiczFlor/RPi-Jukebox-RFID #285: Feature Request: Play only one Track per swipe

Franzformator, 2018-10-23:
I have searched allover in GitHub (this project) but I doesn’t fount a function like this. 
I think it would be nice to have an option to play only one track of a folder after swiping the card for this folder. 
The Use-Cases that come to my mind until now are:
-	A Card with funny sounds (e. g. animal sounds). After every swipe another sound is played.
-	Lullaby-Title. After a swipe every night another title is played bevor bedtime.
-	Kids Party. Instead of playing a complete party playlist it could be more exiting to play random hits after every swipe. (A later enhancement of this function could be to configure a shuffle mode for this single folder)

What do you think about this feature?

MiczFlor, 2018-10-24:
I like it - and will leave it open until I get the time to work on it.
Question / use case:
There is a folder with files. And a card that will always play one track from that folder. As I am filling up the folder and at the same time play files from the folder, what is the logic that tracks the playout?

1. **random**: pick one random file from the available files
2. **random don't repeat**: pick a random file and check if that was the last file played in that folder, if not play it, if yes, pick again
3. **playlist next**: store the last played item from the playlist containing all the files. When swiped, play the next item - or start with first again (problem here: if you keep adding, the playlist keeps changing that might cause repeats)
4. **keep track of played files** (most complicated): there is a list of played files which is being erased once all files in the folder have been played. Until then: pick a random file that is not on the "list of played files" add it to the list and play the file

Option 1 and 3 would be the easiest to implement, I think. Followed by 2 and then, with an additional few hours to work it out, number 4.

Is there another way of thinking this through=

Franzformator, 2018-10-24:
Thank you for your fast response! I think in the beginning an easy solution should be fine for most use cases.
From the list of your suggestions I would prefer number 2 (or alternatively number 3). Especially the random function could bring some fun.

Number 4 would be really cool, especially with a random (shuffle) mode. But I for the time being Number 2 would be a great start.

Franzformator, 2018-11-08:
I found a really nice site with sounds: http://www.geräuschesammler.de/haushaltsgeraeusche
This collection is a perfect example for the “One Sound per Swipe” feature.

mawa85, 2018-12-07:
Great idea, I would prefer option 3
My kid could listen to 1 episode of an audio story every evening.

As a bonus it would be amazing if I could limit it to n tracks of a playlist per day :)

aaukt, 2018-12-10:
I implemented a single play mode in this commit https://github.com/aaukt/RPi-Jukebox-RFID/commit/7f6c82a1c45d92494465b8978f2b301c16c5d977 hadn't had time yet to do the changes on the frontend, but the single mode can be activated in folder.conf.

Together with the RANDOM option you can do things like a dice, an animal sound quiz, random funny sounds etc.

Franzformator, 2019-02-28:
Is there anything new on this Topic? Is it possible to move the code from aaukt to the stable branche?

MiczFlor, 2019-03-01:
Hi @Franzformator 
it's merged with `develop` and there are a few new things which need checking. I am not sure when it could be merged with `master`. Having said this: if you find time testing the current `develop` branch, that would help a lot.
Thanks, micz

MiczFlor, 2019-03-01:
Hi @Franzformator @aaukt 
(correcting my last post: it was NOT merged :)
It is merged with `develop` now. I also added the toggle switch to the web UI. It does not behave as intended yet. For reasons I don't get, IF the mpc single is ON and I switch the settings for a folder back to OFF, the folder.conf correctly reads OFF, but the mpc single is not switched off.
If you can, please have a try in the `develop` branch.

MiczFlor, 2019-03-02:
Hi @Franzformator @aaukt 
I think I fixed the loading of folder specific settings (single track play, shuffle content) and possibly increased the speed of the Phoniebox a little.
So far: each time a new playlist was loaded, if called a separate script to check for shuffle and to check for single play. The problem was: this was not called if you stopped or paused a playlist and then hit play again. It was only done when starting a new playlist.
Now I am including for both "start play" actions (new and already loaded playlist) this script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/inc.settingsFolderSpecific.sh
which loads the folder conf and then sets single play on or off, the same for shuffle.
Please have a look, it's in the `develop` branch.

ZyanKLee, 2020-01-18:
@MiczFlor I suggest closing this issue as it seems to be solved

steffakasid, 2020-06-12:
Does this still work in v2.0? I've got a folder with shufflen: on and singlPlay: on but it doesn't get shuffled. I end up playing the same file all the time (which was first played).

s-martin, 2020-06-12:
It should still work, however there were couple of changes for 2.0, e.g. #889, so it's possible that something was broken in the meantime.

steffakasid, 2020-06-12:
The debug.log generally looks good. Any ideas how I could try to verify what could go wrong?

steffakasid, 2020-06-12:
I found out: If single is on, random will be off an there is no shuffle executed. I did a little change in inc.settingsFolderSpecific.sh:
```
    if [ $SINGLE == "ON" ]
    then
		if [ "${DEBUG_inc_settingsFolderSpecific_sh}" == "TRUE" ]; then echo "  # # CHANGING: mpc single on" >> $PATHDATA/../logs/debug.log; fi
	if [ $SHUFFLE == "ON" ]
	then
		mpc shuffle
	fi
        mpc single on
        mpc random off
    else
```

The change is:
```
if [ $SHUFFLE == "ON" ]
	then
		mpc shuffle
	fi
```
I'll finalize ist and make a pull request tomorrow.

Regards
Steffen

steffakasid, 2020-06-12:
I also found out that CURRENTFILENAME in folder.conf is always the wrong filename. But it seems to doesn't matter for the play out.