MiczFlor/RPi-Jukebox-RFID #287: RPi Zero w + Spotify issues and slow

swyer222, 2018-10-23:
Hi all,

I've successfully update my early jukebox build to the newest version as I was keen to try out the Spotify integration. It does work, but the stream breaks off after about 45-60 secs. Not sure if this is just bc of the low specs of the zero w. 

Then again, compared to my prior build, there now is a noticable lag between swipe and play track. So might just be the hardware.

Anyone else experiencing this?

Thanks
C

princemaxwell, 2018-10-23:
Do you have the noticable lag between swipe and play only for local tracks and livestreams?

swyer222, 2018-10-23:
Local and stream

princemaxwell, 2018-10-23:
@swyer222 @MiczFlor 
That's an issue i am working on at the moment.
This has to do with mopidy. There are two possible solutions i am testing. Deactivate ID3 interpretation, so your playlist only shows the filename. This worked for me and play command is fast as in the pure MPD version.
The second is to put the local files into the mopidy database once. This solution is under testing at the moment... i will inform you, which solution will be pull requested by me.

So please eat a Snickers ;-)

princemaxwell, 2018-10-24:
I decided to realize the second option.
This will take 1-2 days to integrate into develop.

The solution will be to put the files into mopidy database so that they not have been scanned for metadata every time you load them.

swyer222, 2018-10-24:
Thanks @princemaxwell 

I've looked into the Spotify playback issue a bit more. I used _top_ to look into CPU load for **mopidy**: 
- Local playback ~20% 
- Spotify playback ~40%
When I open the browser with the jukebox controls, the CPU load for **mopidy** is twofold and playback is breaking off. The lag is also much more noticeable when having the browser controls open. 

Question: Is it possible to use a different instance for the browser controls and run the process with different priority to ensure smooth playback?

MiczFlor, 2018-10-24:
Hi @swyer222 
possibly a lot to ask, but could you try the "non spotify" version again? And see if the latest "non spotify" from the one line script creates the same problem? Ideally use a new SD card to keep the working version intact.
The reason I am asking is because there could be two reasons for this (especially the web app CPU):
1. the spotify mopidy install drains CPU power
2. the new ajax loading in the web app is too CPU intensive
Or, in fact both and what you experience is unrelated but true :)

swyer222, 2018-10-24:
Sure. Will verify when I have another SD Card. My second set burned up :(

princemaxwell, 2018-10-24:
@swyer222 @MiczFlor 
Mopidy is a massive tool, which was designed to integrate many formats. For example, it would be possible to integrate tunein, GoogleMusic, Soundcloud into Phoniebox.
On the other side, Mopidy is the only tool, which provides Spotify integration on Raspberry.

What we have to see is, that the classic Version of Phoniebox uses MPD/MPC. MPD is a protocol, MPC is the command line tool, which controls MPD.
Mopidy is a Software, which can understand the MPD protocol. MPC commands are routed through Mopidy. These routing is just a little bit slower than in the classic version.

BUT: If you have a local audio folder or livestream, Phoniebox creates a M3U, which will be loaded through Mopidy. Because of some old issues in Mopidy, Mopidy routes the M3U loading trough requests and tries to read the file Information (ID3tags) from each file, file by file. This is what takes a long time (a playlist with 20 files can take 10-15 seconds).

The first solution is, to configure the metadata_timeout in the Mopidy.conf and set it to 1ms. Then Mopidy only reads the filenames and is not getting ID3tags because of timeout.

The second and much better solution is, to change Mopidy file handling and change the method from M3U to Local. In my develop version I tested this and local playlist are as fast as in the classic version of Phoniebox (less than 2 seconds) ;-) Therefor I integrated a button for scanning the local file library on Raspberry. The files are scanned and indexed ONCE! The positive result is, that we have ID3tags available for showing in the WebUI.

I think I can pull request a first version later this evening for testing.

And yes, the AJAX loadings are CPU intensive, because we fire many commands over MPC to get all the informations without the need to reload manually.
Perhaps we have to set the refresh rate up?!?!?!

ekorn, 2018-10-24:
While phoniebox used vlc as player, I build a version that worked with www.pimusicbox.com. Speed wise it was ok on a rpi3 but on the my zero it was way to slow. Especially my custom resume feature which used  mpstore and mpload from the mpdtoys package. Also https://f-droid.org/de/packages/org.gateshipone.malp/ did not work with it. Maybe it was because pimusicbox uses a old version of raspbian or the configuration of mopidy was not perfect, in any case I was very happy see the phoniebox switched to mpc and had a much better resume function so I dumped my code.

So I'm how the new version performs.

princemaxwell, 2018-10-25:
@swyer222 
### 1.1.8-beta is live on the DEVELOP branch!

**For everone who wants to test the newest version with some stability and speed optimizations, feel free to do it and give us a feedback please!**

I updated the UPGRADE instructions today, but have to wait that @MiczFlor will merge this into develop.
Until he has done, here is the instruction.

```
cd /home/pi/RPi-Jukebox-RFID
git checkout develop
git fetch origin
git reset --hard origin/develop
git pull
sudo systemctl stop mpd
sudo systemctl stop mopidy

USERNAME=$(sudo grep 'username' /etc/mopidy/mopidy.conf|sed 's/username = //g'|sed 's/"//g'|tr -d "\n")
PASSWORD=$(sudo grep 'password' /etc/mopidy/mopidy.conf|sed 's/password = //g'|sed 's/"//g'|tr -d "\n")
CLIENT_ID=$(sudo grep 'client_id' /etc/mopidy/mopidy.conf|sed 's/client_id = //g'|sed 's/"//g'|tr -d "\n")
CLIENT_SECRET=$(sudo grep 'client_secret' /etc/mopidy/mopidy.conf|sed 's/client_secret = //g'|sed 's/"//g'|tr -d "\n")

sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/mpd.conf.sample /etc/mpd.conf
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/mopidy-etc.sample /etc/mopidy/mopidy.conf
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/mopidy.sample /home/pi/.config/mopidy/mopidy.conf

sudo sed -i 's/%spotify_username%/'"$USERNAME"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_password%/'"$PASSWORD"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_id%/'"$CLIENT_ID"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_secret%/'"$CLIENT_SECRET"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_username%/'"$USERNAME"'/' ~/.config/mopidy/mopidy.conf
sudo sed -i 's/%spotify_password%/'"$PASSWORD"'/' ~/.config/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_id%/'"$CLIENT_ID"'/' ~/.config/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_secret%/'"$CLIENT_SECRET"'/' ~/.config/mopidy/mopidy.conf

echo "classic" > /home/pi/RPi-Jukebox-RFID/settings/edition
EDITION=$(grep 'SPOTinstall' /home/pi/PhonieboxInstall.conf|sed 's/SPOTinstall="//g'|sed 's/"//g'); if [ $EDITION == "YES" ]; then echo "plusSpotify"; else echo "classic"; fi > /home/pi/RPi-Jukebox-RFID/settings/edition

sudo apt-get install libspotify12 python-cffi python-ply python-pycparser python-spotify
sudo rm -rf /usr/lib/python2.7/dist-packages/mopidy_spotify*
sudo rm -rf /usr/lib/python2.7/dist-packages/Mopidy_Spotify-*
cd
sudo rm -rf mopidy-spotify
git clone -b fix/web_api_playlists --single-branch https://github.com/princemaxwell/mopidy-spotify.git
cd mopidy-spotify
sudo python setup.py install

sudo reboot
```

princemaxwell, 2018-10-31:
### EVERYONE WITH PROBLEMS STARTING LOCAL PLAYLISTS HAS TO RUN THIS COMMAND ONCE:

Depending of the location of audiofolder (this is for default location):
`sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/shared/audiofolders/`

### If you have +Spotify Edition (>=1.1.8-beta) you have to SCAN MUSIC LIBRARY
-> Renew your database under Folders & Files.

And please upload this file to fix the upload in the WebUI or wait until @MiczFlor has merged it!
https://github.com/princemaxwell/RPi-Jukebox-RFID/commit/500922f1f768d21e334844ab0f2a047a8cddba90

closed 2020-01-18 by MiczFlor