MiczFlor/RPi-Jukebox-RFID #291: m4b upload not possible by webgui

no-Legend, 2018-10-25:
Hi,

es scheint nun nicht mehr möglich sein zu sein m4b Dateien hochzuladen.
Bis vor kurzem konnte ich diesen Dateityp noch auswählen und hochladen.

Gruß Robert

pgx3d, 2018-10-28:
Hallo Robert

Ich konnte "alle Dateien" anstatt Audiodateien auswählen, dann wurden auch m4b angezeigt. Diese konnte ich so auswählen und hochladen.

Gruss
Manuel

no-Legend, 2018-10-30:
@pgx3d 
Benutzt du Windows?
Ich arbeite auf OSX.

hier mal zwei Bilder.
Eine Datei als m4b nicht auswählbar, da grau.
Sobald mp3 kein Problem.
[image]
[image]

Gruß Robert

pgx3d, 2018-10-30:
Nope, Linux

MiczFlor, 2018-10-30:
Hi @no-Legend 
das ist echt weird. Ich habe keinen Filter beim Upload welche Files unterstützt werden und welche nicht... Hat OSX schon client-seitig Upload Filter eingebaut ? :)

no-Legend, 2018-10-31:
@MiczFlor 
Ich verstehe es auch nicht,
Hatte bis zum Update auf 1.1.7 das problem nicht.
im Firefox auf OSX gibt es das problem nicht. scheint wohl irgendwie an safari zu liegen.
Es scheint heute ein Update für Safari veröffentlich worden zu sein.

no-Legend, 2018-11-01:
@MiczFlor 
auch nach dem update mit safari kein m4b upload möglich

no-Legend, 2018-11-01:
@MiczFlor mit OSX und dem Firefox geht es. Das mit dem Safari versteh ich nicht.
Bis zum update auf 1.1.7 ging es ohne Probleme.
Ganz komisch.

Hab auch einen anderen User Probiert.
Hier ist genau das gleiche Problem.

closed 2018-11-21 by MiczFlor