MiczFlor/RPi-Jukebox-RFID #297: Question: Experience with Hotspot-Service

toblinga, 2018-10-28:
Hi All,
has someone experience with the install and servrice script which is included in this project?
Does someone use this service?
* RPi-Jukebox-RFID/scripts/installscripts/stretch-install-default-HotspotAddOn.sh
* RPi-Jukebox-RFID/misc/sampleconfigs/autohotspot.stretch-default2-Hotspot.sample
* RPi-Jukebox-RFID/misc/sampleconfigs/autohotspot.service.stretch-default2-Hotspot.sample

I would like to have exacly this features (if home wifi not available, create hotspot to be able to connect the mobile phone and have controll :) )

Thanks for your comments.
Cheers

gizmo21, 2018-10-30:
As I understand those scripts above are halted "work in progress" atm: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/225#issuecomment-424314891 - but correct me MiczFlor 

In the same thread I discussed exactly your feature request the other way round:
So not creating a hotspot on RPi but adding the (mobile)-hotspot of phone (or any other hotspot)  to your raspi-config:
It is already implemented in the code in a simple way to add additional (mobile) hotspots to your RPi config via  RPi-Jukebox-RFID/scripts/userscripts/addhotspot.sh
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/225#issuecomment-424531924

I even added a way to launch this addhotspot.sh via webinterface if you navigate to to the still "hidden" page: <yourraspi_IP or raspi_hostname>/userScripts.php
(https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/userScripts.php)
On that page just select "addhotspot.sh" in dropdown and add the parameters: <SSID> <PASSWORD> 
without the brackets into the parameter-box and execute.

This should add right lines into the 
/etc/wpa_supplicant/wpa_supplicant.conf
/etc/dhcpcd.conf
Which you could check afterwards.

In the manual I also mentioned the fully manual way of doing this including option of another staticIP-wifi (instead of standard DHCP) or priorizing SSIDs and also a to set a hostname instead of using IP, cause this is crucial for multiple wifis:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/229/commits/b5301659c93201964548ecdf2157737267d056a1#diff-abf7b7597cd302b334773a55ee96aae5 

----

Please check if adding SSIDs via <yourraspi_IP or raspi_hostname>/userScripts.php works and report back, cause we need some alternative user feedback ;) on this hidden function.

toblinga, 2018-10-30:
Hi, thank you for this replay and explanation.
I've already configured the SSID for my android hotspot, but than i was not able to control the phoniebox / open the webpage.

And how do you add the Hotel SSID via web-ui? Outside my Homewifi i'm not able to open the web-ui therefor no configuration possibility.

gizmo21, 2018-10-30:
Mobile hotspot: First of all check if your RPi did connect to your mobile hotspot at all. Usually can see this in the hotspot preferences of you mobile phone. If you use IP adress to connect to webui you need to check in your mobilephone which DHCP-adress your RPi got and try that new one -  or try the default hostname http://raspberrypi 
That's why using an own host name is useful (see manual.rd)

Hotel-Hotspot / Friends WIFI: yeah you have to connect via mobile hotspot first to add an hotel/friends SSID/password via the webui and after that (perhaps reboot RPi and) disable mobile hotspot  to get the connect to hotel hotspot.
A bit tricky but I guess doable. 

But as seen in manual: be aware of controllable/"hackable" phoniebox in open wifis

gizmo21, 2018-10-30:
@toblinga did you add you mobile hotspot via webUI userScript.php ? Would be nice to hear if that worked.

MiczFlor, 2018-10-30:
Hi, 
I have a rudimentary file in the repo under scripts for a version that is meant to check *if no ssid I know in reach, become hotspot*. I started working on this, but then never really got it to work.
The main concern on my side: I do *not* want this to be core code, because it does not work for all raspberry pis. It works for generation 3 and Zero (which I haven't checked).
In my script is the link to the original article. If somebody wants to give it a try, please be my guest.

closed 2020-01-18 by MiczFlor