MiczFlor/RPi-Jukebox-RFID #298: 1.1.7 Charset problems, playlist not starting

Joebinator, 2018-10-29:
Hi there!

I habe upgraded to v1.1.7 Spotify works like a charm now thank you very much for that.

Now i have encountered some issues with tracks that contain key's like "ü ä ö" etc

The playlist would not start, i changed raspi-config from UTF-8 to ISO8859-1 now SOME lokal albums work but some are still not working :(

Playlists are getting generated and are correct if i swipe the card or start via web interface but nothing is played.

Also if i click the drop down symbol on the album in the web interface no playlist would be expand. Cover is loaded.

If you need more infos, screenshots, let me please know :)

Joebinator, 2018-10-29:
I found out that, if the Folder contains key's with "ü ä ö" etc i causes that problem, if i rename the folder it works.

Not optimal but it works.

princemaxwell, 2018-10-29:
Normally you have to use UTF-8. It's a workaround in the upgrade instruction and install script. Look there, this might be the solution...

Joebinator, 2018-10-29:
Yeah but UTF-8 isn't working for me ISO8859-1 somehow works with the workarounds i have discovered.

princemaxwell, 2018-10-29:
That's curious. For me UTF-8 works with umlauts like äöü.
ISO is not working because of playlist issues.

Try a new install or check the steps for UTF-8 handling. Don't configure it in raspi-config...
Do it on command line...

princemaxwell, 2018-10-31:
### EVERYONE WITH PROBLEMS STARTING LOCAL PLAYLISTS HAS TO RUN THIS COMMAND ONCE:

Depending of the location of audiofolder (this is for default location):
sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/shared/audiofolders/

### If you have +Spotify Edition (>1.1.8-beta) you have to SCAN MUSIC LIBRARY
-> Renew your database under Folders & Files.

And please upload this file to fix the upload in the WebUI or wait until @MiczFlor has merged it!
https://github.com/princemaxwell/RPi-Jukebox-RFID/commit/500922f1f768d21e334844ab0f2a047a8cddba90

closed 2018-11-04 by MiczFlor