MiczFlor/RPi-Jukebox-RFID #299: No reaction when choosing a playlist in the web interface

floppybox, 2018-10-29:
Thank you for all your work with this project!

I have used the one-line-Phoniebox-installation-script with Spotify today and used all the standard settings on a Raspberry Pi 3B +.
I'm using the Raspberry AUX-audio-output and no GPIO buttons so far.

I can use the card-reader, register new cards properly, play Spotify-tracks and Spotify-playlists and upload mp3s.

Now my problem:
When clicking the play-button of a playlist (only one file "World.mp3") in the web app at "Available audio" the site is reloading but nothing else happens. 
No audio output. No playing time to see. 
When using Spotify everything works fine...

Very strange to me....
Hope you can help me.
Any suggestions are welcome!

I have checked the "Audio is not working" guide.
Here are my findings:

cat /proc/asound/modules
 0 snd_bcm2835

sudo nano /usr/share/alsa/alsa.conf
defaults.ctl.card 0
defaults.pcm.card 0

floppybox, 2018-10-29:
sudo cat /etc/mpd.conf | grep playlist_directory
playlist_directory		"/var/lib/mpd/playlists"

I have changed it according a prior comment.
Now:

sudo cat /etc/mpd.conf | grep playlist_directory
playlist_directory		"/home/pi/RPi-Jukebox-RFID/playlists"

Rebooted and uploaded new music.
Nothing changed. No music playing...

princemaxwell, 2018-10-29:
If you have the Spotify version the MPD.conf user not used. All the music is routed through mopidy.

Have you tried the new develop version 1.1.8??
There were some changes to the local file handling...

floppybox, 2018-10-29:
Thank you for your answer!
Yes, I have the Spotify-version.

> Have you tried the new develop version 1.1.8??

No, I used Master
How is it possible to use the new develop version 1.1.8 as a new install? 
Does there exist a one-line-installation-script, too? 
I would like to give it a try.

princemaxwell, 2018-10-29:
You don't have to make a new install.
You can upgrade...

On command line you have to fire up some lines...

### 1.1.8-beta is live on the DEVELOP branch!

**For everone who wants to test the newest version with some stability and speed optimizations, feel free to do it and give us a feedback please!**

I updated the UPGRADE instructions today, but have to wait that @MiczFlor will merge this into develop.
Until he has done, here is the instruction.

```
cd /home/pi/RPi-Jukebox-RFID
git checkout develop
git fetch origin
git reset --hard origin/develop
git pull
sudo systemctl stop mpd
sudo systemctl stop mopidy

USERNAME=$(sudo grep 'username' /etc/mopidy/mopidy.conf|sed 's/username = //g'|sed 's/"//g'|tr -d "\n")
PASSWORD=$(sudo grep 'password' /etc/mopidy/mopidy.conf|sed 's/password = //g'|sed 's/"//g'|tr -d "\n")
CLIENT_ID=$(sudo grep 'client_id' /etc/mopidy/mopidy.conf|sed 's/client_id = //g'|sed 's/"//g'|tr -d "\n")
CLIENT_SECRET=$(sudo grep 'client_secret' /etc/mopidy/mopidy.conf|sed 's/client_secret = //g'|sed 's/"//g'|tr -d "\n")

sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/mpd.conf.sample /etc/mpd.conf
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/mopidy-etc.sample /etc/mopidy/mopidy.conf
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/mopidy.sample /home/pi/.config/mopidy/mopidy.conf

sudo sed -i 's/%spotify_username%/'"$USERNAME"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_password%/'"$PASSWORD"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_id%/'"$CLIENT_ID"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_secret%/'"$CLIENT_SECRET"'/' /etc/mopidy/mopidy.conf
sudo sed -i 's/%spotify_username%/'"$USERNAME"'/' ~/.config/mopidy/mopidy.conf
sudo sed -i 's/%spotify_password%/'"$PASSWORD"'/' ~/.config/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_id%/'"$CLIENT_ID"'/' ~/.config/mopidy/mopidy.conf
sudo sed -i 's/%spotify_client_secret%/'"$CLIENT_SECRET"'/' ~/.config/mopidy/mopidy.conf

# services to launch after boot using systemd
# -rw-r--r-- 1 root root  304 Apr 30 10:07 phoniebox-rfid-reader.service
# 1. delete old services (this is legacy, might throw errors but is necessary. Valid for versions < 1.1.8-beta)
echo "### Deleting older versions of service daemons. This might throw errors, ignore them"
sudo systemctl disable idle-watchdog
sudo systemctl disable rfid-reader
sudo systemctl disable startup-sound
sudo systemctl disable gpio-buttons
sudo rm /etc/systemd/system/rfid-reader.service 
sudo rm /etc/systemd/system/startup-sound.service
sudo rm /etc/systemd/system/gpio-buttons.service
sudo rm /etc/systemd/system/idle-watchdog.service
echo "### Done with erasing old daemons. Stop ignoring errors!" 
# 2. install new ones - this is version > 1.1.8-beta
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-rfid-reader.service.stretch-default.sample /etc/systemd/system/phoniebox-rfid-reader.service 
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-startup-sound.service.stretch-default.sample /etc/systemd/system/phoniebox-startup-sound.service
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-gpio-buttons.service.stretch-default.sample /etc/systemd/system/phoniebox-gpio-buttons.service
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-idle-watchdog.service.sample /etc/systemd/system/phoniebox-idle-watchdog.service
sudo chown root:root /etc/systemd/system/phoniebox-rfid-reader.service
sudo chown root:root /etc/systemd/system/phoniebox-startup-sound.service
sudo chown root:root /etc/systemd/system/phoniebox-gpio-buttons.service
sudo chown root:root /etc/systemd/system/phoniebox-idle-watchdog.service
sudo chmod 644 /etc/systemd/system/phoniebox-rfid-reader.service
sudo chmod 644 /etc/systemd/system/phoniebox-startup-sound.service
sudo chmod 644 /etc/systemd/system/phoniebox-gpio-buttons.service
sudo chmod 644 /etc/systemd/system/phoniebox-idle-watchdog.service
# enable the services needed
sudo systemctl enable phoniebox-idle-watchdog
sudo systemctl enable phoniebox-rfid-reader
sudo systemctl enable phoniebox-startup-sound
sudo systemctl enable phoniebox-gpio-buttons

echo "classic" > /home/pi/RPi-Jukebox-RFID/settings/edition
EDITION=$(grep 'SPOTinstall' /home/pi/PhonieboxInstall.conf|sed 's/SPOTinstall="//g'|sed 's/"//g'); if [ $EDITION == "YES" ]; then echo "plusSpotify"; else echo "classic"; fi > /home/pi/RPi-Jukebox-RFID/settings/edition

sudo apt-get install libspotify12 python-cffi python-ply python-pycparser python-spotify
sudo rm -rf /usr/lib/python2.7/dist-packages/mopidy_spotify*
sudo rm -rf /usr/lib/python2.7/dist-packages/Mopidy_Spotify-*
cd
sudo rm -rf mopidy-spotify
git clone -b fix/web_api_playlists --single-branch https://github.com/princemaxwell/mopidy-spotify.git
cd mopidy-spotify
sudo python setup.py install

sudo reboot
```

MiczFlor, 2018-10-30:
Hi @princemaxwell 
if you give the thumbs up, I will merge the develop 1.1.8 with the master branch. My feeling is (and has been for some time) that the `develop` is not being used for testing. And I wouldn't either, after all, why challenge a running system.
Therefore, if we do merge to master, and announce it as beta, people can decide if they do want to upgrade.
What do you think?
My feeling is that it will create a small avalanche of bug reports - and then a quick rush for fixes and then some peace and quiet, a plateau, until the next major addition to the code base :)

princemaxwell, 2018-10-30:
@MiczFlor
Thumbs up!
My tests with develop have shown no problems.
Please change the develop entries in the UPGRADE.md to master.

Would it be better to write a little script for updaters? The number of update steps is high...

Nordwig, 2018-10-31:
Hi im new here. First thanks to everyone for the work on this cool Projekt.

It seems i have the same problem. Spotify works just fine but local stored files dosent start to play.
Spotify plays a song after i have started a local file ( via RFID or Web interface), the playback stops and the light grey Box with song title time and playlist ect. disappears.

I upgraded from a working 1.1.7 non Spotiy Version  to 1.1.8 Beta via the one Link install.
I kept my config, RFID and audiofiles.
After that i tried to upgrade again with the one link install nothing changed.

princemaxwell, 2018-10-31:
Please provide some additional information.
Can you check the content of /settings/playlist_folder_path and /etc/mopidy/mopidy.conf (be sure to remove you Spotify informations!!!)

Have you restored config files after update?

Post the contents here.

Nordwig, 2018-10-31:
the content of /settings/playlist_folder_path
`/home/pi/RPi-Jukebox-RFID/playlists`

the content of /etc/mopidy/mopidy.conf
```
[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy

[logging]
config_file = /etc/mopidy/logging.conf
debug_file = /var/log/mopidy/mopidy-debug.log

[local]
enabled = true
media_dir = /home/pi/RPi-Jukebox-RFID/shared/audiofolders
excluded_file_extensions =
  .conf
  .jpg
  .txt
  placeholder

[file]
#enabled = true
metadata_timeout = 1

[m3u]
playlists_dir = /home/pi/RPi-Jukebox-RFID/playlists
default_encoding = UTF-8
default_extension = .m3u

[audio]
output = alsasink
mixer_volume = 30

[mpd]
hostname = 0.0.0.0

[http]
hostname = 0.0.0.0

[iris]
country = DE
locale = de_DE

[spotify]
enabled = true
username = 
password = 
client_id = 
client_secret =
#bitrate = 160
#volume_normalization = true
#private_session = false
#timeout = 10
#allow_cache = true
#allow_network = true
#allow_playlists = true
#search_album_count = 20
#search_artist_count = 10
#search_track_count = 50
#toplist_countries =
```
This is after i updated from 1.1.7 no Spotify to 1.1.8Beta with Spotify with the one link installer
`cd; rm stretch-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/stretch-install-spotify.sh; chmod +x stretch-install-spotify.sh; ./stretch-install-spotify.sh`

I havent restored any config files after the update.
I choosed the fresh install with default file folder

I hope it helps

princemaxwell, 2018-10-31:
Ok, thats looking great.
Lets look a little bit deeper. Please look into your folder /home/pi/RPi-Jukebox-RFID/playlists.
There have been m3u files, which will be created when you press the play button beside a local folder in the Web UI or swipe a card.
Take the name of the folder, which is not working in the web UI and take look if there is a m3u file for it.
It should have the name "foldername.m3u". Open it with an editor like notepad or notepad++ and paste the content (if its not to much) here.

Then open a command line and type in line after line
```
mpc clear
mpc load "foldername"
mpc play
mpc current
mpc playlist
```

Copy the content and paste it here.

Nordwig, 2018-10-31:
Name of the folder is "Eisbrecher" file name is "Eisbrecher - Polarstern.mp3" 
in /home/pi/RPi-Jukebox-RFID/playlists is "Eisbrecher.m3u" contend is: `local:track:Eisbrecher/Eisbrecher%20-%20Polarstern.mp3`

the command line:
```
pi@raspberrypi:~ $ mpc clear
volume: 30%   repeat: off   random: off   single: off   consume: off
pi@raspberrypi:~ $ mpc load "Eisbrecher"
loading: Eisbrecher
pi@raspberrypi:~ $ mpc play
volume: 30%   repeat: off   random: off   single: off   consume: off
pi@raspberrypi:~ $ mpc current
pi@raspberrypi:~ $ mpc playlist
```

There was no sound.

princemaxwell, 2018-10-31:
### EVERYONE WITH PROBLEMS STARTING LOCAL PLAYLISTS HAS TO RUN THIS COMMAND ONCE:

Depending of the location of audiofolder (this is for default location):
`sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/shared/audiofolders/`

### If you have +Spotify Edition (>1.1.8-beta) you have to SCAN MUSIC LIBRARY
-> Renew your database under Folders & Files.

And please upload this file to fix the upload in the WebUI or wait until @MiczFlor has merged it!
https://github.com/princemaxwell/RPi-Jukebox-RFID/commit/500922f1f768d21e334844ab0f2a047a8cddba90

princemaxwell, 2018-11-01:
@Nordwig
Great! Have you updated the manageFilesFolders.php? Otherwise new uploaded files don't get the right permission rights...

closed 2018-11-04 by MiczFlor