MiczFlor/RPi-Jukebox-RFID #301: Jump to a certain time „on a line“ in WebApp

harryliebteuch, 2018-10-29:
I would be happy if there is a timeline in the WebApp in which I could Jump to a certain. That's great for long files.
I mean something that every player has downstairs like z.b. the VLC, or Youtube

princemaxwell, 2018-10-30:
I think this will not possible.
The fact is that we have not really a music player. We control a music protocol by command line. The only thing we can is seek seconds or minutes for- and backwards.

What could be possible is a timeline bar in the web UI. But without control feature.

Franzformator, 2018-10-31:
Would it may be possible to get a drop-down-list with certain time steps to jump to a specific section?

For example: 
-	If the audiofile is 10 min. long, the list could display the time in 10 sec steps (10, 20, 30, ...)
-	If the audiofile is 5 min. long the list could display the time in 5 sec. steps (5, 10, 15, …)

princemaxwell, 2018-10-31:
We cannot seek to a specific section.
What we can do is seeking by percentage.

The only way to seek for- and backwards is this option

seek [+-][<HH:MM:SS>] or <[+-]<0-100>%>
Seeks by hour, minute or seconds, hours or minutes can be omitted. If seeking by percentage, seeks within the current song in the specified manner. If a "+" or "-" is used, the seek is done relative to the current song position. Absolute seeking by default.

Please have a look here for all available commands:
https://linux.die.net/man/1/mpc

princemaxwell, 2018-11-05:
@harryliebteuch 
We could use the seek command of MPC for that.
Its possible to seek by seconds or by percentage.

so the easiest way would be to add a swipe action for 5%, 10%, 20% and so on.
Thats what @Franzformator suggested. As longer the file, as longer the jump.

What did you mean @MiczFlor ?