MiczFlor/RPi-Jukebox-RFID #302: lighttdp is not working

Moostante, 2018-10-30:
Hi, thank you for this great project!

I have installed the Phoniebox on a Raspberry 3B+ using the latest master branch, following the installation instructions in INSTALL-stretch.md step by step (not the one line install command), using headless installation via ssh.

Now I encountered a problem with lighttpd and am unable to find a solution on my own. I would be very glad if you could help me!
I have comleted and rechecked all the steps up to:

sudo service lighttpd force-reload

This line always produces the output:
Job for lighttpd.service failed because the control process exited with error code. See "systemctl status lighttpd.service" and "journalctl -xe" for details.

pi@phoniebox:~ $ journalctl -xe 
--
-- Unit lighttpd.service has failed.
--
-- The result is failed.
Oct 30 07:51:28 phoniebox systemd[1]: lighttpd.service: Unit entered failed state.
Oct 30 07:51:28 phoniebox systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Oct 30 07:51:29 phoniebox systemd[1]: Stopped Lighttpd Daemon.
-- Subject: Unit lighttpd.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit lighttpd.service has finished shutting down.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Start request repeated too quickly.
Oct 30 07:51:29 phoniebox systemd[1]: Failed to start Lighttpd Daemon.
-- Subject: Unit lighttpd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit lighttpd.service has failed.
--
-- The result is failed.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Unit entered failed state.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Failed with result 'exit-code'.

pi@phoniebox:~ $ systemctl status lighttpd.service
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-10-30 07:51:29 UTC; 9min ago
  Process: 1408 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=255)

Oct 30 07:51:28 phoniebox systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Oct 30 07:51:29 phoniebox systemd[1]: Stopped Lighttpd Daemon.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Start request repeated too quickly.
Oct 30 07:51:29 phoniebox systemd[1]: Failed to start Lighttpd Daemon.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Unit entered failed state.
Oct 30 07:51:29 phoniebox systemd[1]: lighttpd.service: Failed with result 'exit-code'.

marcohorstmann, 2018-10-30:
Can you post here please the lighttpd.conf? maybe an permissions error I found online as a possible reason. What says "ls -lisa /etc/lighttpd/lighttpd.conf"?

Moostante, 2018-10-31:
pi@phoniebox:~ $ ls -lisa /etc/lighttpd/lighttpd.conf
181538 4 -rw-r--r-- 1 root root 1107 Oct 19 21:31 /etc/lighttpd/lighttpd.conf

pi@phoniebox:~ $ cat /etc/lighttpd/lighttpd.conf
server.modules = (
        "mod_access",
        "mod_alias",
        "mod_compress",
        "mod_redirect",
)

# server.document-root        = "/var/www/html"
# Changed by Maike
server.document-root        = "/home/pi/RPi-Jukebox-RFID/htdocs"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"

Thank you for helping!

MiczFlor, 2018-11-09:
Hi @Moostante 
please try to reinstall using the latest one-line-install script. The current version is 1.1.8-rc2
The install doc for manual install is behind the latest code. Sorry about that.

closed 2018-11-09 by MiczFlor