MiczFlor/RPi-Jukebox-RFID #308: Max98357a Volume control

foonky, 2018-11-01:
Hi,

I try to Build a Jukebox with the mentioned Mono amp.
But it is always at 100% Volume.

In a Volumio Forum they Talk about Software Mixer.

Has someone got. ist working

The MAx98357 doesnt support Hardware volume control so you must set Vollumio to: 
Mixertype=software
MixercontrolName=software volume control

Ardios, 2018-11-01:
@foonky 

try this:

sudo nano /etc/mpd.conf

under audio_output uncomment the mixertype line and change "hardware" to "software"

then save and reboot

MiczFlor, 2018-11-21:
Hi @foonky,
Assuming your silence means „thanks, working now“, I‘ll close this issue? :)

closed 2018-11-21 by MiczFlor

foonky, 2018-11-21:
Actually i received a new one yesterday.The old one was broken. It is working Fine now