MiczFlor/RPi-Jukebox-RFID #309: Powerbank without interruption while charging

olsins, 2018-11-01:
I connected my pi to a Powerbank which supports the function to charge and running the pi at the same time. Unfortunately there is a short interruption of the power supply when I disconnect the Powerbank from the external charging plug. 
The solution is described in an heise article 

https://books.google.de/books?id=HAo4DwAAQBAJ&lpg=PA81&ots=Ug1Rj_N3IY&dq=Eine%20einfache%20USB-Powerbank%20macht%20den%20Raspberry%20Pi%20mobil.%20Doch%20ein%20Dauerbetrieb%20ist%20damit%20nicht%20m%C3%B6glich%3A%20Beim%20An-%20oder%20Abstecken%20des%20Ladekabels%20bricht%20die%20Versorgungsspannung%20so%20weit%20zusammen%2C%20dass%20der%20Raspi%20neu%20bootet.%20Elkos%20und%20Goldcaps%20f%C3%BCllen%20die%20L%C3%BCcke.&hl=de&pg=PA81#v=onepage&q&f=false

they connect an Elko 18.000 6.3 Volt to the pins 4 and 6.

Now the question - how can I apply this solution if I still want to use an external swith to turn the pi on and off.

Right now the switch connects to GPIO3 (PIN5) and GND (PIN6). If I have an Elko connected to 4 and 6 at the same time they would be a circuit 465 with 6.3 Volts every time I press the button.
Can I do that or will I break my pi ? Does anyone know how I could manage to interrupt the connection 4 and 6 every time I connect 5 and 6?
Or any other solution to make this work?
Thanks a lot for any help!

Andco7, 2018-11-01:
,> To make it even harder I also want to connect an external knob for sound control. This post
#267
also includes 5 6.

The mentioned post uses GPIO 5 and GPIO 6 which is Pin 31 and Pin 33 on the board.
I use the Shutdown/Boot Buttons through GPIO 3 (Pin 5) and GND (Pin 6) in combination with the rotary-encoder and it works.

olsins, 2018-11-01:
I mixed pin an GPIO - thanks for the hint about the volume control -  so the problem will only be for the Elko 18.000

olsins, 2018-11-02:
More thoughts on the Elko Solution: Since PIN4 is 5V Power and PIN6 is GND -  is there a problem if PIN3(GPIO2) is connected toPIN 4 at the same time.  I mean obviously the board is designed to connect 5V to go through GPIO2 for 2 Seconds. The Elko runs with 6.3 Volt - do I have to put in something (resistor) to make it 5V? Or will it be fine?

kaoz3000, 2018-11-13:
Die 6,3V stehen für die Belastbarkeit des Elkos. Die Spannung bleibt aber bei 5 Volt, wenn man ihn nur mit 5 Volt auflädt.

xCite1986, 2019-01-06:
this one works great
https://www.amazon.de/gp/product/B01422TC14/

derOger, 2019-01-15:
Hello, can you please send some details how you have wired the bank exactly? I want to use the same solution maybe. Very regards.

wdmnn, 2019-01-16:
I can just second what xCite said. I'm using the same Powerbank and you can plug/unplug power without affecting the running PI.
Tested the 10k Version as well but wasn't working.
What exactly do you want to know regarding wiring? You just power the PI via the Powerbank and use GPIO 5/6 for Power On/Down (assuming you don't use an additional hardware like OnOffShim to powerOff the Pi completely).

derOger, 2019-01-16:
When I use a USB-powered speaker and RFID-reader and maybe an OLED-Display and halt the Pi will the power cinsumption drop to zero? I thought that the Pi will still consume power.
I think about using a OnOffShim because of that.

xCite1986, 2019-01-16:
hey @derOger 
i've conncted the input-side from powerbank with the 2 cables which are included to my computer and the raspberry pi with a short microusb to the output-side. the 

that works pretty well but 2 things:

- if the bank gets empty and if only the inner port was re-connected with my computer the raspberry gets really unstable and doesnt boot up again. it looks like the inner port alone couldnt provide the current to charge the powerbank and reboot the raspberry at the same time. so at least the outer port works well - but i think both input-connectors was better for the micro-sdcard :)

- if i disconnect the powerbank from my computer and the pi was running there was no problem and did not interruping PI. when i shut them down with holding GPIO3 (PIN5) and GND (PIN6) 2 seconds together the pi gets off and a minute later the powerbank turns the secondary side off. so i couldnt turn on the raspberry anymore with the shutdown-button and i have tu push the putton on the powerbank to turn them on first, then the pi cames up again. i hope i find a soltion for this [(maybe like that)](https://forum.pjrc.com/threads/28624-USB-Battery-Bank-Prevent-Shut-Down?s=a7e833595dc194c9f58408c5df4265e6&p=131247&viewfull=1#post131247)

sorry for my english, it was'nt the best but I hope it is understandable explained

closed 2024-05-09 by s-martin