MiczFlor/RPi-Jukebox-RFID #311: no file exensions on uploaded files

dtrv, 2018-11-03:
Hi. I observed a strange behavior using Chrome browser on an android smartphone: All uploaded files with mp3 as file extension (.mp3) get stored without its extensions. So 'mysong.mp3' becomes 'mysong'. Files with mp4 as extension seems not to be affected. With desktop version of Chrome browser the upload works as expected.
Tested on two different Smartphones (ok, both Android and both Chrome browser).

Can't say if it's an Android feature/bug, Chrome related or an issue with the manageFilesFolders.php

Any help/suggestions welcome!

MiczFlor, 2018-11-04:
Hi @dtrv 
that's a weird one. We had a similar thing with Safari on OSX not allowing to upload certain file types, depending on their file ending.
I can't test it. Once you fix or understand it, please post the wisdom here, to share with others.
All the best, micz

dtrv, 2018-11-05:
Tested with Firefox on Android. Upload works as expected. So it's a problem with Chrome and the file dialog for audio files of Android. Strange that nobody else seems to have this problem ... I'll try to upload by not using the file selector for audio files. Will report after testing ...

Fast search brought me to [Html File Input on Chrome for Android missing extension and mime type](https://stackoverflow.com/a/30739278/4073745) on SO which is from 2015.

princemaxwell, 2018-11-05:
@drtv
Thanks for your tests. That's really crazy. Seems to be a chrome only problem on Android.
So... Buy an iPhone ;-)

closed 2018-11-09 by MiczFlor