MiczFlor/RPi-Jukebox-RFID #312: No mp3 Playback, but Spotify works

rebhuhnweg, 2018-11-04:
Hello,

I have already done the installation 2 times. Both times exactly the same problem. What works already:

1. playing Spotify Streams
2. the start sound during booting
3. output either on USB or onboard jack
4 The RFID Reader
5. GPIO e.g. for SHitdown

No matter which Mp3 I upload and whether I want to start it via RFID or the web interface, nothing happens. I don't see a cover (correctly copied) either.

What am I doing wrong ? I have everything ready. Software, hardware, cases, buttons, cards. Only the most important thing (for which I built it for my daughter) is to play MP3.

Could anyone help me pls ?

Ronnie

MiczFlor, 2018-11-04:
Hi Ronnie @rebhuhnweg 
Could you please post the results here that you get when you run this in the Phoniebox terminal?
~~~
cat /home/pi/RPi-Jukebox-RFID/settings/version
cat /home/pi/RPi-Jukebox-RFID/settings/Audio_Folders_Path
cat /etc/mpd.conf | grep music_directory
~~~
The latest version for Spotify had a couple of fixes regarding the local audio play, so if 
~~~
cat /home/pi/RPi-Jukebox-RFID/settings/version
~~~
does not read `1.1.8-rc1` please try the latest install.
A quick fix could be:
~~~
sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/shared/audiofolders/
~~~
and then try again. There was an issue with uploaded files that could not be read my `mpd` because of the access rights. The latest install should have fixed that.
I am also pinging @princemaxwell here because he is the dude that knows most about the Spotify integration in the community (hands down).
P.S.: I like the `shitdown` you are mentioning. That must be a freudian slip / typo given the frustration you might have experienced :)

rebhuhnweg, 2018-11-05:
What I found out yesterday was that every time I shoot mopidy via SSH and then start again it works. After I then followed the tip of @MiczFlor and gave with chmod right on the Shared/Audiofiles folder it runs now :)

Great, thank you very much.

closed 2018-11-05 by rebhuhnweg

FFKite, 2018-11-09:
Hey, 
a big thank to MiczFlor. 
I have the same problems. 

i have this returns:
`s_Path cat /etc/mpd.conf | grep music_directory 
cat: cat: No such file or directory
cat: cat: No such file or directory
music_directory		"/var/lib/mpd/music"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.`

can you help me?

FFKite, 2018-11-09:
I think you should use an
`s_Path cat /etc/mpd.conf | grep music_directory 
cat: cat: No such file or directory
cat: cat: No such file or directory
music_directory		"/var/lib/mpd/music"
music_directory are changed.
symbolic links outside of the configured music_directory.
symbolic links inside of the configured music_directory.` element here instead.

olsins, 2018-11-09:
I had it all running then I changed my Samba password. After that the Youtube download did not work anymore -  so I reinstalled everything. Same behavior. The shutdown button does not work anymore and nothing is played over the interface. Something must have been changed through the last update.

olsins, 2018-11-09:
I reinstalled everything without the Spotify Option and it works again. Only the encoder still makes trouble (other thread)

princemaxwell, 2018-11-10:
@olsins
Did you use the installer?

Have you checked your permissions?

sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/shared/audiofolders/

Have you scanned your local music library after uploading the files? Files that are not scanned cannot be played.

You find the scan option under folders & files.

princemaxwell, 2018-11-10:
If you didn't scan your library in folders & files, that must be the reason for the Spotify version. The files must be in the mopidy database. You have to scan library every time you upload new files...

MiczFlor, 2018-11-10:
Hi @princemaxwell 
would it make sense to habe a „one off“ service like the startip sound for the linrary scan?
That would allow to say „after adding new files reboot - or eescan the library“ some people might reboot without RTFM and it would work. 

Is the scan requirement in the manual?
All the best, micz

alanrodriguezdotme, 2018-11-23:
Hello! I'm having this same problem. I'm able to get Spotify working w/o any issues, but I can't get MP3s to play. I used the install script, changed permissions to 777 for all folders in /home/pi/RPi-Jukebox-RFID/shared/audiofolders/, and I scanned the folder through the web app. I'm still not able to get MP3s to play. I only have 1 folder named `dance` with 3 songs, none of which have any whitespace.

Any suggestions?

alanrodriguezdotme, 2018-11-24:
Nevermind! I uploaded another song through the web UI, and it works!