MiczFlor/RPi-Jukebox-RFID #313: Resume often fails

mmally, 2018-11-05:
Hi,

I just set up the latest version 1.1.8-rc1 with spotify and found that the resume function fails quite often.

I found two problems:
1. ELAPSED is save as float in folder.conf e.g. ELAPSED="136.854", but mpd seek only seems to supports seconds as integer. I fixed it by converting it to integer when passing it to seek (in resume_play.sh line 115)

2. querying mpd status often does not return the time, elapsed,... although a mp3 is playing.
You can replicate it, if you do a second swipe:
1st swipe mpd status returns elapsed, 2nd swipe it is missing.
3rd swipe it works again, 4th fails again and so on.

I hope someone can help me fix this issue, bcause I am building the Phoniebox for he birthday of my son.

Best Regards,
Michael

MiczFlor, 2018-11-06:
Hi @mmally 
can you give us more experience with this issue? 
I am running an install without spotify and there I can not reproduce the issue. Might have to do with the mopidy / spotify packages. @princemaxwell , since you are the wizard of OS for this install, can you reproduce this?

princemaxwell, 2018-11-06:
@MiczFlor
Can do the tests on weekend, I am on seminar this week...

mmally, 2018-11-07:
I can confirm that it works fine using Phoniebox without Spotify integrated

shinebar1001, 2018-11-09:
Same here - when I activate Resume the Playback fails on the first swipe with "incorrect Arguments"

```
Nov  9 13:30:30 phoniebox python2[560]: volume: 30%   repeat: off   random: off   single: off   consume: off
Nov  9 13:30:30 phoniebox python2[560]: loading: Michael Jackson - Best of Playlist
Nov  9 13:30:30 phoniebox python2[560]: volume: 30%   repeat: off   random: off   single: off   consume: off
Nov  9 13:30:30 phoniebox python2[560]: OK MPD 0.19.0
==> Nov  9 13:30:30 phoniebox python2[560]: ACK [2@0] {seek} incorrect Arguments <===
Nov  9 13:30:30 phoniebox python2[560]: OK
```

The second swipe Restarts the Playlist from the beginning:
```
Nov  9 13:30:37 phoniebox python2[560]: volume: 30%   repeat: off   random: off   single: off   consume: off
Nov  9 13:30:38 phoniebox python2[560]: loading: Michael Jackson - Best of Playlist
Nov  9 13:30:38 phoniebox python2[560]: volume: 30%   repeat: off   random: off   single: off   consume: off
Nov  9 13:30:38 phoniebox python2[560]: OK MPD 0.19.0

```

MiczFlor, 2018-11-09:
Hi @mmally @shinebar1001 
let me call @princemaxwell again, he is the Spotify mastermind and (by now) might have seen all the things that are glitches in the system.
If you figure this out or have figured it out already, please post your wisdom here, so I can include it in the manual / documentation.

princemaxwell, 2018-11-10:
Will check this on evening/night.

StefanSimon, 2018-11-15:
Hi together,
I was faced to the same problem, but only with some files. For these files it restarted the playlist every time, when I tried to resume on them.
In my case the problem was caused by the filename. The files had an apostrophe ( ‘ ) in their names.
Maybe this could be the reasons for others.

marvin78, 2019-02-13:
Is there something new about this topic?

closed 2019-06-21 by MiczFlor