MiczFlor/RPi-Jukebox-RFID #314: Can't choose subfolders from symlink to USB stick

varac, 2018-11-05:
I am using an USB stick and have symlinked linked folders on it to `~/RPi-Jukebox-RFID/shared/audiofolders` like this:

```
$ mount | grep Media
/dev/sda1 on /home/pi/Media type ext4 (rw,noatime,data=ordered)

pi@musicbox:~/RPi-Jukebox-RFID/shared/audiofolders $ ls -al
total 24
drwxrwxr-x 5 pi   www-data 4096 Nov  5 21:14 .
drwxrwxr-x 4 pi   www-data 4096 Nov  4 22:49 ..
lrwxrwxrwx 1 pi   pi         30 Nov  4 22:03 Audiobooks -> /home/pi/Media/Audiobooks
drwxr-xr-x 3 pi   pi       4096 Nov  5 21:14 Compilations
-rwxrwxrwx 1 root root       98 Nov  4 10:32 folder.conf
lrwxrwxrwx 1 pi   pi         26 Nov  4 22:02 Music -> /home/pi/Media/Music/
```

In the home screen I can browse all subfolders of i.e. Audiobooks, see this screenshot where it shows correctly 3 subfolders:

[image]

But I can't choose any subfolder below Audiofiles in `/cardRegisterNew.php`:

[image]

`Compilations` is a real folder beneath `~/RPi-Jukebox-RFID/shared/audiofolders` and I can select subfolders from the card register page though.

`www-data` can read all subdirs so I think this is not a permission problem:

```
root@musicbox:~# sudo -u www-data ls -al /home/pi/Media/Ayla/Audiobooks/
total 20
drwxrwxr-x  5 pi www-data 4096 Jun 19 15:37 .
drwxrwxr-x  4 pi www-data 4096 Jun 19 12:42 ..
drwxrwxr-x  3 pi www-data 4096 Jun 19 14:54 Janosch
...
```

Any idea how to register a new card to symlink subfolders ?

I'm using the `Version-1.1.5-dynamicRFIDreader` branch.

varac, 2018-11-05:
Weird - I symlinked the whole `audiofolders` dir to the mountpoint of the usb stick (`/home/pi/Media`) and with this it works.

MiczFlor, 2018-11-06:
Might be a access rights issue. Following symlinks via http daemon often requires that the parent also has access rights granted for the webserver.

varac, 2018-11-09:
Ok, closing for now.

closed 2018-11-09 by varac