MiczFlor/RPi-Jukebox-RFID #315: Switch on GPIO not possible

webjochen, 2018-11-06:
I have completely reinstalled the raspberry.
strech image installed and then
with the oneline command jukebox installed without spotify.
I can play music, rfid cards work
but i can not turn on the gpio buttons in the web interface. that always worked for me right away.
the function "Show Cover" is without function for me.

MiczFlor, 2018-11-06:
What version are you running? The 1.1.8-rc1?

webjochen, 2018-11-06:
yes, I installed it yesterday with oneline command

MiczFlor, 2018-11-06:
What does this return?
~~~
/bin/systemctl status phoniebox-gpio-buttons.service
~~~
I changed the services from phoniebox all to have names starting with `phoniebox`. Possibly that created a problem. Calling @marcohorstmann if he could help here (he was the one triggering the idea to have them all named the same - possibly he checked if that worked :)

webjochen, 2018-11-06:
`phoniebox-gpio-buttons.service - Phoniebox GPIO Buttons Service
   Loaded: loaded (/etc/systemd/system/phoniebox-gpio-buttons.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-11-06 11:00:50 CET; 47s ago
  Process: 533 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py (code=exited, status=203/EXEC)
 Main PID: 533 (code=exited, status=203/EXEC)

Nov 06 11:00:50 jukebox systemd[1]: phoniebox-gpio-buttons.service: Unit entered failed state.
Nov 06 11:00:50 jukebox systemd[1]: phoniebox-gpio-buttons.service: Failed with result 'exit-code'.
Nov 06 11:00:50 jukebox systemd[1]: phoniebox-gpio-buttons.service: Service hold-off time over, scheduling restart.
Nov 06 11:00:50 jukebox systemd[1]: Stopped Phoniebox GPIO Buttons Service.
Nov 06 11:00:50 jukebox systemd[1]: phoniebox-gpio-buttons.service: Start request repeated too quickly.
Nov 06 11:00:50 jukebox systemd[1]: Failed to start Phoniebox GPIO Buttons Service.
Nov 06 11:00:50 jukebox systemd[1]: phoniebox-gpio-buttons.service: Unit entered failed state.
Nov 06 11:00:50 jukebox systemd[1]: phoniebox-gpio-buttons.service: Failed with result 'exit-code'.`

marcohorstmann, 2018-11-07:
Is the file "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py" existing? 
what happens when you run the command "sudo /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py" from your shell? Any error message?

I just upgraded my daughters box to the 1.1.8-rc1. No system for fresh installation available.

webjochen, 2018-11-07:
the file is available

pi@jukebox:~ $ sudo /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py

sudo: /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py: Befehl nicht gefunden

pi@jukebox:~ $

marcohorstmann, 2018-11-07:
Please run sudo again without :

webjochen, 2018-11-07:
pi@jukebox:~/RPi-Jukebox-RFID/scripts $ ls
daemon_rfid_reader.py           playlist_recursive_by_folder.php
deviceName.txt                  playout_controls.sh
gpio-buttons.py                 Reader.py
helperscripts                   Reader.pyc
idle-watchdog.sh                RegisterDevice.py
inc.readArgsFromCommandLine.sh  resume_play.sh
inc.writeFolderConfig.sh        rfid_trigger_play.sh
installscripts                  rotary-encoder.py
ky040.py                        shuffle_play.sh
MFRC522.py                      userscripts

pi@jukebox:~/RPi-Jukebox-RFID/scripts $ sudo gpio-buttons.py
sudo: gpio-buttons.py: Befehl nicht gefunden

pi@jukebox:~/RPi-Jukebox-RFID/scripts $

MiczFlor, 2018-11-07:
Did you switch the GPIO on in the Raspberry Pi config?

webjochen, 2018-11-07:
I have executed the following command
it works now

`sudo chmod +x /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py`

but the do not show cover is still without function

marcohorstmann, 2018-11-07:
which filename and location did you use for covers?

webjochen, 2018-11-08:
Web interface:
Do you want to show covers on the albums and playlists on the main page?

this does not work for me
The cover on the main page is always displayed
no matter what I choose

MiczFlor, 2018-11-09:
Hi @webjochen 
nice catch with the gpio buttons. I can't trace back when that sneaked into the install, because I had no complains in earlier versions. 
Included your fix in the latest release: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/322/files

closed 2018-11-09 by MiczFlor