MiczFlor/RPi-Jukebox-RFID #317: Issus Upgrade from Version 1.1.7 to 1.1.8-beta

woll-knaeuel, 2018-11-06:
Hello
I tried to upgrade my Phoniebox version from 1.1.7 to 1.1.8-beta and ran into the following problems.

```
Package libffi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libffi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libffi' found
c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
 #include <ffi.h>
```
solved with: sudo apt-get install libffi-dev

2nd try ended with:

```
warning: no previously-included files matching '__pycache__/*' found anywhere in distribution
build/temp.linux-armv6l-2.7/spotify._spotify.c:492:28: fatal error: libspotify/api.h: No such file or directory
 #include "libspotify/api.h"
                            ^
compilation terminated.
error: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

looks like a missing libspotify12 or libspotify-dev and indeed a couple of lines above the package installation fails

```
root@raspberrypi:~/mopidy-spotify# sudo apt-get install libspotify12 python-cffi python-ply python-pycparser python-spotify
Reading package lists... Done
Building dependency tree        
Reading state information... Done
E: Unable to locate package libspotify12
E: Unable to locate package python-spotify
```

Probably I am not the only one who runs into these problems so I just wanted to let you know.

Best regards

MiczFlor, 2018-11-06:
Hi @woll-knaeuel 
thanks for the info. So far it's been very quiet regarding the spotify version of 1.1.8-rc1. Did you install from scratch?

MiczFlor, 2018-11-06:
Hi @woll-knaeuel 
my bad: you upgraded. Not sure if it would be better to install with the one-line-script - after a backup of course. There have been a lot of tweaks since the 1.1.8-beta

woll-knaeuel, 2018-11-06:
Ok, I will wait for the final and install from scratch than.

MiczFlor, 2018-11-09:
Just released rc-2 for 1.1.8
We are close to releasing 1.1.8 stable (... well, you know what that means: less bug reports than ever...)
Possibly this is good enough to try an install from scratch and see what you get?

closed 2018-11-09 by MiczFlor