MiczFlor/RPi-Jukebox-RFID #319: Remove placeholder file from audiofolders and shortcuts

varac, 2018-11-06:
I removed `audiofolders/` and `shortcuts/` and replaced them with symlinks to a usb drive.
Now they show up as deleted in my git status:

        deleted:    shared/audiofolders/placeholder
        deleted:    shared/shortcuts/placeholder

I guess the `placeholder` files are just there so you can checkin both directories into git. Can we removed them ?

MiczFlor, 2018-11-06:
HI @varac 
that is correct. They are only there so that the empty folders can be part of the github repo

closed 2018-11-06 by MiczFlor

varac, 2018-11-07:
My proposal was to remove them (and not check in `audiofolders/` and `shortcuts/` into git), so symlinking doesn't show up as git changes.
And maybe create them dynamically from the install script.