MiczFlor/RPi-Jukebox-RFID #320: Track scripts/gpio-buttons.py in git

varac, 2018-11-07:
Currently, `misc/sampleconfigs/gpio-buttons.py.sample` gets copied to `scripts/gpio-buttons.py` by the `stretch-install-spotify.sh` install script.
But then it's git ignored by the `.gitignore` script, which makes it impossible to check it into version control if doing manual changes.

I propose to remove it from `misc/sampleconfigs/` and put it directly into `scripts/gpio-buttons.py`.

marcohorstmann, 2018-11-07:
that is in my opinion no good option. Depending on which hardware you have you maybe need to change this file. Somebody had some days ago posted here that he need to change the script because of I think hifiberry was using some of the GPIO pins we are using in the default script. you will get problems with git and maybe every time you need to change the file after a update.

varac, 2018-11-09:
I agree. Best solution would be to move the pin layout to a config file like others, and then checkin the actual script. Closing for now.

closed 2018-11-09 by varac