MiczFlor/RPi-Jukebox-RFID #321: How to add a custom playlist ?

varac, 2018-11-08:
Scenario: I want to create a "mix", a playlist that referes to different tracks from different artists. The tracks already exist in `audiofolders` i.e.:

- audiofolders/Artist-A/track07.mp3
- audiofolders/Artist-B/track02.mp3
- audiofolders/Artist-C/track01.mp3

How would I create such a playlist manually and link a new RFID tag to it, without duplicating the mp3 (the easy way would be to copy all tracks in a new directory, but that wasting disk space).

MiczFlor, 2018-11-09:
Hi @varac 
I have plans to build one (1) custom playlist. The idea to use this would then be:
* in the web app next to the file listings there is an *add to master playout* or the like
* clicking this will add the file to the list
* there will also be a master playlist management, allowing to remove files or clear the list entirely

By doing this, the files will not be copied, but the master list can use the absolute file path on the system.
Timeline (i.e. when?) is a bit difficult to estimate. I hope to do this before Christmas.
I believe it is a great thing to add to the Phoniebox. At the same time, the Phoniebox does already so much more than the core functionality that I am very proud of all the community ideas and contributions that made its way into the code. And the master playlist is more for the grown ups than the kids :) (I am just saying this to protect myself from promising deadlines...)

MiczFlor, 2018-11-09:
Hi @varac 
I gave this a little thought and wrote down the following instructions that might give you a head start on making this feature work:

## Prep the code manually
In the file `rfid_trigger_play.conf.sample` I added the following lines:
~~~
# Custom Playlist
CMDPLAYCUSTOMPLS="%CMDPLAYCUSTOMPLS%"
~~~
If you want to make this work for your install, you need to add this to the file in the running system
`settings/rfid_trigger_play.conf` and add the ID of the RFID card instead of the `%CMDPLAYCUSTOMPLS%` (keep the " in place).

Then I squeezed three new lines of code into `scripts/rfid_trigger_play.sh` (here they are in context):
~~~
$DISABLEWIFI)
    $PATHDATA/playout_controls.sh -c=disablewifi
    ;;
$CMDPLAYCUSTOMPLS)
    $PATHDATA/playout_controls.sh -c=playlistaddplay -v="PhonieCustomPLS" -d="PhonieCustomPLS"
    ;;
$STARTRECORD600)
    #start recorder for -v seconds
    $PATHDATA/playout_controls.sh -c=startrecord -v=600			             
        ;;
~~~
New are these three:
~~~
$CMDPLAYCUSTOMPLS)
    $PATHDATA/playout_controls.sh -c=playlistaddplay -v="PhonieCustomPLS" -d="PhonieCustomPLS"
    ;;
~~~
Add them as well.

## Create the playlist

Create an empty file in the `playlists` folder (I assume that is the place where you have the playlists, the standard install) with the name `PhonieCustomPLS.m3u`
~~~
touch /home/pi/RPi-Jukebox-RFID/playlists/PhonieCustomPLS.m3u
~~~

Inside this file you can add a list of files with the absolute path to the actual files, which could look like this:

~~~
file://home/pi/RPi-Jukebox-RFID/shared/audiofolders/abc-lied/abc-lied.mp3
~~~

The `file:` might need three slashes, I can't quite remember. So it might be 
~~~
file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/abc-lied/abc-lied.mp3
~~~

And add all the file paths you want.

Then the swipe might work. I give this a 10% chance of success. Would be curious to hear if this worked and if not: how you made it work. Then it could possibly become a hack for the next release to get a UI later.

Possibly you need to create a folder in the `shared/audiofolders` path by the name `PhonieCustomPLS` as well. As I said, I haven't tried this at all :) I doubt that shuffle and or resume will work. But it might just play the content of the playlist.

Keep us posted with your results!

varac, 2018-11-13:
@MiczFlor Thx for the instructions! However, from what I got going over them, these are still limited to only *one* custom playlist right ? This is already an improvement, but not what I'm looking for. I'd like to put multiple custom .m3u playlist files somewhere and link them to different rfid tags.
I'm not urging to implement this soon, it just would be a great feature for phoniebox, even without UI.

closed 2020-01-18 by MiczFlor