MiczFlor/RPi-Jukebox-RFID #324: Empty playlist generated for folders including '&'

varac, 2018-11-09:
When an audio folder contains a `&`, the auto-generated playlist is empty.

varac, 2018-11-09:
```
pi@musicbox:~/RPi-Jukebox-RFID $ ls -al shared/audiofolders/Q\ \&\ A/                                                                                                                    
total 8
drwxr-xr-x  2 pi pi       4096 Nov  9 23:34 .
drwxrwxr-x 10 pi www-data 4096 Nov  9 23:38 ..
-rw-r--r--  1 pi pi          0 Nov  9 23:34 1.mp3
-rw-r--r--  1 pi pi          0 Nov  9 23:34 2.mp3

pi@musicbox:~/RPi-Jukebox-RFID $ ./scripts/playlist_recursive_by_folder.php folder="Q\ \&\ A" list='recursive'                                                                           
PHP Notice:  Undefined variable: folders in /home/pi/RPi-Jukebox-RFID/scripts/playlist_recursive_by_folder.php on line 78
PHP Warning:  Invalid argument supplied for foreach() in /home/pi/RPi-Jukebox-RFID/scripts/playlist_recursive_by_folder.php on line 78
```

MiczFlor, 2018-11-21:
Hi @varac
Did you find a hack to solve this? I would like to include it in the code base. All the best, micz

varac, 2018-11-21:
@MiczFlor No, I didn't. I tried but am too unfamiliar with the code.
My workaround was to rename all albums and files by removing the `&` :o

closed 2019-03-08 by MiczFlor