MiczFlor/RPi-Jukebox-RFID #325: GPIO ports for buttons and rotory seems to conflict

lathspell, 2018-11-10:
The https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume uses GPIO 5, GPIO 13 and GPIO 16.

Then there is https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/misc/sampleconfigs/gpio-buttons.py.sample which uses among others:

    vol0 = Button(13,pull_up=True)
    volU = Button(16,pull_up=True)

I now fear that both scripts try to initialize and register callback functions on the same GPIO 13/16.

marcohorstmann, 2018-11-10:
Yes you're right. I messed up the GPIO PIN/BCM Encoding in another post.
Please use this little modified version of gpio-buttons.py where I disabled all volume controls.
```
#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for additional info: 
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)

shut = Button(3, hold_time=2)
#vol0 = Button(13,pull_up=True)
#volU = Button(16,pull_up=True)
#volD = Button(19,pull_up=True)
next = Button(26,pull_up=True)
prev = Button(20,pull_up=True)
halt = Button(21,pull_up=True)

shut.when_held = def_shutdown
#vol0.when_pressed = def_vol0
#volU.when_pressed = def_volU
#volD.when_pressed = def_volD
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt

pause()
```

lathspell, 2018-11-12:
Aren't there enough GPIO pins so that the button-script and the rotary-script could coexist? I would like it if they both get starter by default and the docs just say, "connect the rotray to pin x,y,z or a button to xx,yy or both if you want, it will just work..."

marcohorstmann, 2018-11-19:
Probable yes but because everyone uses other hardware it will be difficult to build one fits all script.

lathspell, 2018-11-21:
But Phoniebox is a pretty specialized project. Most users will only have a subset of prev/next/pause, shutdown, sleep timer, folder prev/next, rotary volume, button volume, mute. That's what, 15 pins or so? The Raspberry3 has > 20 if I remember correctly so there would be plenty left for "special purposes".

Of course adjusting the python scripts it easy but it would be even cooler if one could just plug a button at the right point and have the functionallity without having to remember to adjust the mapping at every update. It would aid for the slogan "no soldering needed and no programming knowledge needed"

marcohorstmann, 2018-11-21:
but sometimes you have special hats what blocks several pins. I don't know I I should build it that it fits for everyone. In my opinion impossible. On every update you don't need to change gpio-buttons.py because it will not be overwritten.