MiczFlor/RPi-Jukebox-RFID #326: Mopidy: failed to add some tracks

MTam86, 2018-11-10:
I have created a new installation today. But unfortunately I get an error message immediately when I try to play something on Spotify.
**Mopidy failed to add some tracks**

Server: Connect
Spotify: Authorized

Thanks and Greetings
Malte

tomraithel, 2018-11-16:
I have the same problem. I followed these steps:
1. Launch phoniebox webapp
2. Go to "spotify" page
3. Search for song and hit "play"
4. Getting "Mopidy failed to add some tracks"  as a result

princemaxwell, 2018-11-17:
From here we cannot support the way over the "spotify" button. This button is only to take a look into the web interface and if the server is running fine.
Phoniebox controls spotify and local files on a different way.

closed 2019-03-08 by MiczFlor