MiczFlor/RPi-Jukebox-RFID #330: PhonieBox just refuses to play my music! Tried everything....please help!!

guardianin, 2018-11-13:
Hi. So I spent the whole day yesterday and half the day today trying to get PhonieBox running successfully. But it just refuses to play my files! Neither do the files play from the web player, nor with the RFID card! Detailed explanation below along with environment.

1. Raspberry Pi 3B+
2. 3A adapter (no power issues whatsoever even with all USB ports used)
3. 32GB MicroSD card
4. External powered HDD connected via USB, formatted as ext4
5. 125Khz RFID reader and matching cards
6. Logitech Bluetooth Adapter
7. Creative SoundBlaster Live external soundcard
8. Latest Raspbian Stretch (GUI)

Here is what I tried (multiple times):

1. Fresh install of Raspbian Stretch
2. Configure everything using raspi-config
3. Update system including firmware completely and restart
4. External HDD confirmed mounted automatically
5. All options for external HDD set in fstab - UUID=uuid here /media/pi/pimusic ext4 user,rw,nofail 0 0
6. Additional global permissions given to /media/pi with user pi for read/write access (777 permission)
7. Install PhonieBox using the 2nd one line script with Spotify integration
8. Specified /media/pi/pimusic as the location for audio files
9. All other options configured accordingly
10. Registered RFID reader correctly
11. Reboot

After reboot, I am able to open the web app. My test folder of music also appears, with all the audio files within. However the files just refuse to play. Clicking play in the web app just reloads to page, and the play icon does not change to pause to indicate something is playing. Even touching a registered card to the reader does not invoke playback. I have also tried a Spotify playlist, same results. The LatestID.txt file always gets updated with the last card. The auto-created shortcut files, folders etc also appear in the right places as mentioned.

I have tried all possible combinations in mpd.conf, Audio_iFace_name, asound.conf and wherever else as mentioned in the manual and troubleshooting articles. I've tried all this with the built-in soundcard of the Pi, my Creative USB soundcard, and my Logitech Bluetooth adapter. For the Logitech adapter, I have tried with the built-in BT as well as a USB BT dongle.

But my music just plain refuses to start playing!! I really don't know what else to do.....please help!

maikJ123, 2018-11-14:
Same day, same problem.
When scanning through the logfiles it looks like it tries to open a wrong folder.

cat error.log | grep -v "Notice" | grep -v "file_get_contents(../settings/Latest_Played_File)"

2018-11-13 07:16:57: (mod_fastcgi.c.2543) FastCGI-stderr: PHP message: PHP Warning:  file_get_contents(/home/pi/Music/0007235046): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 9

While home/pi/Music/ is the correct parent folder, 0007235046 is not a folder name but the RFID that should link to it.

Possible that the RFID-folder replacement is broken in the recent version?

guardianin, 2018-11-14:
Ok, I've got it up and running!! FINALLY!!! 👍 :-D

Turns out that I had to force audio output in raspi-config to the headphone jack! Also, the Spotify version does not seem to work at least for me (for now), so I went with the regular version.

Now that I have it up and running, will test it using another SD card for music from an external sound card or Bluetooth. And the Spotify version as well.

Really awesome software though! I knew I was only doing something wrong (just didn't know what) as many people have it up and running. Oh well, good learning experience as I can navigate in the Linux CLI much better now!

closed 2018-11-16 by guardianin