MiczFlor/RPi-Jukebox-RFID #331: Some folders don't play!

guardianin, 2018-11-15:


gusmax777, 2018-11-16:
Hi!
I have similar issues and also tried a lot to solve. But until now, the files only work when being played directly from my SD-Card not from the external drive. Can you explain how linked the harddrive to the media folder? Via ln or did you change the folder value in the settings? Do the files play if you copy them to your SD-Card?
Cheers, Thomas

guardianin, 2018-11-16:
For playing from an external drive you have to do a few things.

Create a mount point for the drive in the OS
Edit the fstab file with the required options so that it mounts automatically on startup
Change the Audio_Folders_Path file to point to the new mount point
Change the folder location in the mpd.conf file.

gusmax777, 2018-11-16:
Hi!
Sure, I am aware of this process, but the result is exactly the same. Some folders play and some not. As soon as you have found a solution, I would be thankful to know.

guardianin, 2018-11-17:
@princemaxwell
Using the Classic Edition, version 1.1.8 RC2.

guardianin, 2018-11-19:
Hi. Any workaround or solution for this issue?

princemaxwell, 2018-11-19:
Can you post some examples?
Some foldernames...

It could be possible that folders with special characters are not playing. But this problem should only be with the +Spotify edition, because folder and file handling changed there.

The classic edition is still using the old routine with MPD.

princemaxwell, 2018-11-21:
Yeah, great! It was my first intention.

@MiczFlor can you try to fix this issue for MPD version?
I will check this for the Spotify edition.

johkoenig, 2018-12-09:
@guardianin: I had similar problems with some mp3 were playing while others weren't. Problem was solved by running "mpc update" everytime after I uploaded some stuff. Now, even the crazyest characters don't break anything.

@MiczFlor: Awesome project, btw. We're setting up a box right now for you 2y daughter for christmas. I do the tech part, which took me less than an hour!

RogoTe, 2018-12-16:
Another problem is folder names with & ampersand. Clicking in the WebUI does not work because the folder is handed over as unmasked string in the URL and so & is recognized as next parameter.

MiczFlor, 2018-12-17:
Hi @RogoTe @johkoenig @guardianin @gusmax777 thanks for catching this bug, There has been a slight improvement in the "special chars" universe with this commit:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/382
It's part of `develop`, does that solve (almost) everything?

johkoenig, 2018-12-18:
@MiczFlor Unfortunately, I cannot do any testing before beginning of next year. I only have one raspi and that one will be in use from christmas on. 
I will try to check as soon as possible, though.

gizmo21, 2018-12-20:
[offtopic tip]
Win32diskimager (or similar for Linux) and a second SD-card gets you a "second" testing-system quite fast - in case you have sparetime left in the evenings where kids should perhaps sleep ;) 

https://sourceforge.net/projects/win32diskimager/

But I know that live OS is no good for testing, that's why I reverted kids SD back to 1.0.0 as is was the most stable for me atm.

mr-gosh, 2018-12-26:
> @guardianin: I had similar problems with some mp3 were playing while others weren't. Problem was solved by running "mpc update" everytime after I uploaded some stuff. Now, even the crazyest characters don't break anything.

I can confirm that normal looking mp3s are not played like these:

```
raspberrypi:/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Musik/Type O Negative/Bloody Kisses $ ls -lah
insgesamt 57M
drwxrwxrwx 2 pi pi 4,0K Dez 26 21:38 .
drwxrwxrwx 5 pi pi 4,0K Dez 23 22:55 ..
-rw-rw-rw- 1 pi pi 570K Aug 12 19:00 01 - Machine Screw.mp3
-rw-rw-rw- 1 pi pi 7,6M Aug 12 19:04 02 - Christian Woman.mp3
-rw-rw-rw- 1 pi pi 9,5M Aug 12 19:09 03 - Black No. 1 (Little Miss Scare-All).mp3
-rw-rw-rw- 1 pi pi 961K Aug 12 19:09 04 - Fay Wray Come Out and Play.mp3
-rw-rw-rw- 1 pi pi 2,9M Aug 12 19:10 05 - Kill All the White People.mp3
-rw-rw-rw- 1 pi pi 4,2M Aug 12 19:12 06 - Summer Breeze.mp3
-rw-rw-rw- 1 pi pi 3,1M Aug 12 19:13 07 - Set Me On Fire.mp3
-rw-rw-rw- 1 pi pi 339K Aug 12 19:13 08 - Dark Side of the Womb.mp3
-rw-rw-rw- 1 pi pi 6,1M Aug 12 19:15 09 - We Hate Everyone.mp3
-rw-rw-rw- 1 pi pi 9,2M Aug 12 19:18 10 - Bloody Kisses (A Death in the Family).mp3-rw-rw-rw- 1 pi pi 1,7M Aug 12 19:19 11 - 3.0.I.F..mp3
-rw-rw-rw- 1 pi pi 4,8M Aug 12 19:23 13 - Blood & Fire.mp3
-rw-rw-rw- 1 pi pi 5,3M Aug 12 19:24 14 - Can’t Lose You.mp3
-rw-rw-rw- 1 pi pi 929K Dez 23 22:52 cover.jpg
-rwxrwxrwx 1 pi pi   98 Dez 26 21:38 folder.conf
```

mpc update did the trick
This should be added to a daily cron or something...

mr-gosh, 2019-01-04:
I don't use the spotify edition!
I suggest inotify could be used to monitor filesystem changes and trigger "mpc update" when changes are detected... - I'll see if I have time to test that setup - for now i integrated it into cron...

closed 2020-01-18 by MiczFlor