MiczFlor/RPi-Jukebox-RFID #332: Incorrect Documentation (MPD)

nlammel, 2018-11-15:
Hi,
thanks for the cool project, but I figured out some issues in the documentation. Just tried the installation as defined in the Wiki. So I could install everything, but I had issues with starting the playback of any music. 

At the end the problem was identified as a wrong setting within the mpd.conf file and the setting of the playlist_directory. The documentation stated that it should be set to "/tmp", but only with setting it as follows made it work: 

playlist_directory              "/home/pi/RPi-Jukebox-RFID/playlists"

Furthermore I also needed to set 
auto_update "yes" (as written) 
auto_update_depth "10" (missing)

otherwise I had to always call mpc update to load new files uploaded into the shared folder.

I used raspbian stretch with mpd 0.19.21-1 and tried to follow step by step your description. Maybe something to improve the already really good documentation.

MiczFlor, 2018-11-18:
Hi @nlammel 
nice catch. Thank you. I will include this asap.

MiczFlor, 2018-11-23:
I added this now, both in the wiki and the default mpd conf for the one line install. Thanks again:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/344

closed 2018-11-23 by MiczFlor