MiczFlor/RPi-Jukebox-RFID #335: push two button for a reset

ddbkva, 2018-11-18:
Hello!
Is there a possibility that if you press two buttons for example 
power (gpio 3) and volume up (gpio 16) press this and the raspberry pis do a restart ??
is that possible ?

marcohorstmann, 2018-11-23:
Hi @ddbkva,

yes that is possible. I build this with next and prev buttons (only buttons I habe on my devbox)
and a little bit research in 0,5-1h this evening.

But now it's time for bed. I added here the content of my gpio-buttons.py. Please try it and provide Feedback.

```
#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call

# 2018-11-23
# Added the function to reboot phoniebox when pressing next and  previous button at the same time.

# 2018-10-31
# Added the function on holding volume + - buttons to change the volume in 0.3s interval
#
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for additional info:
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_reboot():
    check_call("./scripts/playout_controls.sh -c=reboot", shell=True)

def def_volU():
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)

def def_both_pressed():
    if prev.is_pressed and next.is_pressed:
        check_call("./scripts/playout_controls.sh -c=reboot", shell=True)
    elif prev.is_pressed:
        check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)
    elif next.is_pressed:
        check_call("./scripts/playout_controls.sh -c=playernext", shell=True)
    else:
        print("Error")

shut = Button(3, hold_time=2)
vol0 = Button(13,pull_up=True)
volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(19,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(26,pull_up=True)
prev = Button(20,pull_up=True)
halt = Button(21,pull_up=True)

shut.when_held = def_shutdown
vol0.when_pressed = def_vol0
volU.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a ra$
volU.when_held = def_volU
volD.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower t$
volD.when_held = def_volD
#next.when_pressed = def_next
#prev.when_pressed = def_prev
halt.when_pressed = def_halt

next.when_pressed = def_both_pressed
prev.when_pressed = def_both_pressed

pause()

```

Franzformator, 2018-11-23:
This sounds nice!

Maybe this solution makes it possible to realize my old request #168 

My idea was to enable and disable Wifi with a simultaneous press of some buttons.

marcohorstmann, 2018-11-23:
Hi @franzformator,
your request is a little more complicated because I need to extend the main playout controls file. In the current version it has only commands for enablewifi and disablewifi. I need to implement a togglewifi command than this would work. I will take a look the next days to implement this.

marcohorstmann, 2018-11-23:
@Franzformator your request was sucessfully added.
I updated my PR #306 and added a new commit. There is an updated playout_controls.sh to implement the programming logic and a new file gpio-buttons-with-twobuttonshortcut.py.sample.
In this sample config file I used Volume Up and Volume Down button pressed at the same time.
The script allows  you to choose which action will be performed. I added shutdown and togglewifi
as commands but other commands e.g. shutdown are possible too.

So this script is universal for @Franzformator and @ddbkva requests.

This is the commit: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/306/commits/4336936998fc6299bf8e1f38a3c8457a33c2c214

Hopefully @MiczFlor has a good start in his new job and has some time to merge this PR into devel.

closed 2020-01-18 by MiczFlor