MiczFlor/RPi-Jukebox-RFID #337: Poor audio quality when using a power bank

ChristinaVF, 2018-11-18:
Hi,
I just recently built my Phoniebox. To improve the audio quality and eliminate the cracking and hissing noise, I use a USB sound card (from Sabrent). 
But when I use one power bank both for the RPi and the speakers (Trust Leto 2.0), the hissing and cracking noise is back. 

Do I need to use separate power supplies for the RPi and the speakers or is the problem with the USB hub? Or am I the only one experiencing this and using another power bank might solve the issue? Right now I have a Mi Power Bank 10400mAh?

0s1r1s, 2018-11-25:
I have the same problem. 
Just found [this related Issue ](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/341)

MiczFlor, 2018-11-26:
Hi @ChristinaVF @0s1r1s 
thanks for helping each other out :) I will close this, assuming it's solved?

closed 2018-11-26 by MiczFlor