MiczFlor/RPi-Jukebox-RFID #338: Wifi setup does not work in stretch

tikeyknax, 2018-11-18:
I have used the one-line install script to install the phoniebox 1.1.8-rc2. During the installation, I used the option to set up the wifi. However, that did not work, i.e. afterwards my wifi did not connect after a reboot of the raspberry pi. It worked, after I added the following lines to my wpa_supplicant.conf:
```
    update_config=1
    country=AT
```

According to www.rasperrypi.org/forums/viewtopic.php?t=196809, this has changed in stretch. I don't know if this is a general issue. If so, it might be something which should be added to the script and maybe also in the wiki.

hwmrocker, 2018-11-24:
happend to me too
seems like this line is the problem ```    sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs//wpa_supplicant.conf.stretch-default2.sample /etc/wpa_supplicant/wpa_supplicant.conf
```
there is no `wpa_supplicant.conf.stretch-default2.sample` in the repo

MiczFlor, 2018-11-24:
Hi @hwmrocker @tikeyknax 
true, there is not file with such a name. I have to check where that got lost! stay tuned.

closed 2020-01-18 by MiczFlor