MiczFlor/RPi-Jukebox-RFID #341: CPU activity in speaker

rebhuhnweg, 2018-11-20:
Hi,

I have the problem that I hear the CPU activity in the boxes as "quiet" rattling and crackling. Clearly recognizable when CPU activity is. The electrical structure of my box:

1 USB power supply. At the USB cable I soldered the power of the PI as well as the power for the small boxes (those from the shopping list of the project). The RFID reader gets its power from the Pi's USB port. (I had the boxes originally also at the Pi thought then however that I get it away in such a way).

The problem is completely gone when I connect the boxes via a completely separate USB power supply, but that's no solution :(

Do you know the problem, and does anyone have an idea ?

Greeting from northern Germany

Ronnie

MiczFlor, 2018-11-21:
Hi @rebhuhnweg
not just CPU but also wifi activity on the same ssid in general :) to my knowledge this can only be fixed with a soundcard like the hifiberry listed in the readme file. 
I file it under „21st century vinyl sound“ and lived with it for over a year now. :)
All the best, micz

deMerl, 2018-11-21:
Hi @rebhuhnweg 
I had the same issue. Noises were so loud you couldn´t use the Box. Solved this issue with a "Ground Loop Isolator" like this one: [Entstörfilter](https://www.amazon.de/gp/product/B01LX0H29W/ref=oh_aui_detailpage_o09_s00?ie=UTF8&psc=1). Noises are gone completely.
Could maybe also solve it with a USB Soundcard but the sound of the PI 3B+ audio jack is okay for me.
Greez, deMerl

lonjame, 2018-11-22:
The Ground Loop Isolator solved the issue for me

HenningWesterkamp, 2018-11-22:
If you are using a seperate Amplifier on the same powersource as the Pie this is a usual problem. In my case there was noise in the speaker even if the Audio cable between the pi and the amp was not installed.

Try using seperate powersources for the amp and the pi.

marcohorstmann, 2018-11-23:
For PAM8403 amp based setups I created pull request #306

tobiasnix, 2018-11-23:
The Ground Loop Isolator solved the issue for me too

MiczFlor, 2018-11-23:
I added this to the "special hardware" section on the front page, please have a little check if this is accurate:
https://github.com/MiczFlor/RPi-Jukebox-RFID#special-hardware

closed 2018-11-25 by MiczFlor

orangecoding, 2021-06-29:
Just fyi, the link to the "Entstörungsfilter" doesn't work anymore.

MiczFlor, 2021-06-29:
> Just fyi, the link to the "Entstörungsfilter" doesn't work anymore.

Ok, why is that?

dekiesel, 2021-09-16:
> > Just fyi, the link to the "Entstörungsfilter" doesn't work anymore.
> 
> Ok, why is that?

It seems it got removed from amazon.de