MiczFlor/RPi-Jukebox-RFID #349: Issue trying to play a Song/Album from Spotify

phoenichts, 2018-11-25:
Hello,

sorry for bringing up this topic again, even if it was closed in issue #18 a few days ago.

The day before yesterday I installed the Jukebox with Spotify to a fresh installed raspbian stretch lite (without a desktop) using the 1 line install script (thus I'm running V1.1.8-rc2). I'm able to play local content via an external USB-soundcard and according to the iris-page I'm connected to spotify (I doesn't use a Facebook-Login). If I add a track or a album to the Jukebox, the index-page displays the cover of the related album or track (I regard this as a kind of proof that there must be a connection between my Jukebox and Spotify) but if I hit "play" on the webapp to play any content from Spotify, nothing happened execpt a kind of reload from the page. 

I'm not sure if there is a connection between the two facts, but I additionally have the same problem as described in issue #326.

I've no further experience with Linux, hence I'm, a little bit stranded.

I assume that an option is available that allows to log whats happens if I hit play, but neither I do know where to find it nor am I familiar with the (possible) error messages. 

I would be very grateful if some provides me help at this point.

Thanks: Christian

MiczFlor, 2018-11-26:
Hi @phoenichts 
I need to pass this on to @princemaxwell who is the driving force behind Spotify. I will add his response to the Spotify integration doc. 
All the best and welcome to the Phoniebox community, micz

phoenichts, 2018-11-26:
Hi MiczFlor,

thank you. I tried the installation a second and a third time today after recognising that rc3 is out. Unfortunately this doesn't change anything: Local sound works, Spotify does nothing except adding a nice cover on the page and adding a directory with 4 files :(

I'll hope for a feedback from princemaxwell.

BTW: the "oneliner" produces at least two error messages (checked it two times):
[image]

cheers: Christian

princemaxwell, 2018-11-26:
@phoenichts
I am on a seminar this week and can take a look on weekend.
Have you created the web API secrets for Spotify?

I didn't understand what you mean, which files does Spotify create in what directory?

phoenichts, 2018-11-26:
@princemaxwell 
Hi and thanks for your help! A look at the weekend would be great!

I created the secret and the client ID via https://www.mopidy.com/authenticate/ and added them into the install script. My Spotify profile displays mopidy and Iris extension as beeing authorised.

When I had a look into _/etc/mopidy/mopidy.conf_ I found the proper information (except the pw, but I assume that this is correct). _~/.config/mopidy/mopidy.conf_ did not contain any information that it should contain according to the manual install instruction (https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/SPOTIFY-INTEGRATION.md#configure-mopidy). I added them, but this changed nothing... I don't know if this was useful or not.

With regard to the files: If I assign an album or a single track to a card ID, a new directory is created that contains the cover.jpg, the folder.conf, spotify.txt and title.txt. The link in spotify.txt either to the album or to the track is correct.

Best regards and enjoy your seminar: Christian

phoenichts, 2018-11-27:
Hi all,

got it: This issue was caused by a combination of script error and layer 8 error!

The first problem was that the install-script added all credentials needed to connect to Spotify except my password. When I had a look into the *.conf-files I assumed that %password% is a variable. After adding my real password it worked!

By the way: now also the behaviour as described in issue #326 has gone.

Cheers: Christian

princemaxwell, 2018-11-27:
Can you explain why the script didn't add your password correctly?

Normally the script should work fine...

phoenichts, 2018-11-27:
@princemaxwell 

Unfortunately no.

As I wrote: I tried the installation 3 times and the problem that I could not play any content from Spotify was always the same. But I would assume that I make a mistake if no one else has oder had this problem...

Christian

petschni, 2018-11-30:
@phoenichts the current version of the install script generates the following line for the password:

```password = xyzzy%password%```

Removing the `%password%`placeholder and restarting solved the problem for me.

princemaxwell, 2018-11-30:
I will check the install script...

MiczFlor, 2018-12-02:
Hi @phoenichts @petschni 
I can not reproduce / understand how you got this weird half-replaced string in the config file. Did you use the one line install script? Hm. The replacement is here in the code:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/b93d083af2c5265f39f86bfcb54b4d12a85bd115/scripts/installscripts/stretch-install-spotify.sh#L629
To understand better what's going on in the system, I created a "report script", see here for details:
https://github.com/MiczFlor/RPi-Jukebox-RFID#reporting-bugs
The idea is to get more inforamtion about the system.
If this issue is resolved for you, please report here what you did do and then close the thread. Sorry for being vague and puzzled. Thanks, micz

petschni, 2018-12-03:
@MiczFlor I fixed the issue manually for me and had a look at the code but have also no idea no idea why the replacement did not work. (I used the install script.)

woll-knaeuel, 2018-12-05:
Fresh installation from scratch Raspbian GNU/Linux 9.6 (stretch) with Plus edition (feat. Spotify integration) using the one line install script. Except of the already above posted errors (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/349#issuecomment-441811325) everything went smoothly.
Startsound confirmed that the audio should working, adding one mp3 file and one spotify album pressed the play button and... nothing. No audio output. I do not know why, but the audio output problem was produced by a default setting file for mopidy (sudo nano ~/.config/mopidy/mopidy.conf). After replacing this one with the one posted here (https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/SPOTIFY-INTEGRATION.md) at least the mp3 file could be played and heared.
The spotify album is still not be playable. Any suggestions?
[image]
Is there any logfile of mopidy I can check? The credentials have been set correctly in my case to /etc/mopidy/mopidy.conf.

TM4G1, 2018-12-06:
> Hi all,
>
> got it: This issue was caused by a combination of script error and layer 8 error!
>
> The first problem was that the install-script added all credentials needed to connect to Spotify except my [...]

Hi Christian,

where exactely did you add your password? I am facing the same issue. Did a fresh installation today. Mopidy also states that I got authorized, but I cannot play any content from spotify. 

Best regards
Tom

phoenichts, 2018-12-06:
@MiczFlor @TM4G1 

I changed the two files that are described under https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/SPOTIFY-INTEGRATION.md#configure-mopidy

Afterwards Spotify and Mopidy worked.

Cheers: Christian

TM4G1, 2018-12-06:
> @MiczFlor @TM4G1
> 
> I changed the two files that are described under https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/SPOTIFY-INTEGRATION.md#configure-mopidy
> 
> Afterwards Spotify and Mopidy worked.
> 
> Cheers: Christian

Thanks... Checked the second file as well, looks good, but does still not work. Simply no playback of any spotify content. May I generate a new client id and certificate and try again.

andisger, 2018-12-08:
Bei mir lag es daran, dass er mopidy Iris nicht installiert hatte mit dem one line install command.
nach dem ich das manuell nachgeholt habe (gemäß der manuellen Anleitung), lief es.

Zapatito, 2018-12-30:
Hi All,

I also had some trouble to get to run Spotify. After looking into this thread, I checked the mopidy configs as linked by phoenichts:
[https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/SPOTIFY-INTEGRATION.md#configure-mopidy]
I found that the Spotify password was missing in the first file (/etc/mopidy/mopidy.conf). There was no spaceholder. The spot behind "password:" was just blank. In the second config-file I found the correct PW.
My version is 1.1.8, installed with the one line command.

The second thing I found was a mistake I made when assigning the URI to the phoniebox-RFID-tags:
I copied the entire link that I got from the spotify-web-player.
For example: spotify:album:https://open.spotify.com/album/1Y620MsEFRKkindjPq4IvN
(The album title and cover were shown in the phoniebox-web-app. Therefore I thought that I had it linked correctly.)
Instead it works like this: spotify:album:1Y620MsEFRKkindjPq4IvN

Hope the feedback is sort of helpful.

Best regards!

closed 2020-01-18 by MiczFlor