MiczFlor/RPi-Jukebox-RFID #352: resume not working

SimonChelkowski, 2018-11-27:
Error description runs over points 1. to 11.. After this an extract of the "Analytics_AfterInstallScript.sh" follows

1. Buttons in webinterface do not change anything. No resume can be activated via webinterface. The issue is related to #110 which apparently is closed although the problem described there still exists

2. Via ssh I can enable resume for the folder that is currently played from by running
	pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ./resume_play.sh -c=enableresume -v=D
Note that the value (-v=...) specified in the command is not relevant, whatever I type there only the folder.conf of the folder that is currently played is modified. This is not in-line with what is described in the header of the "resume_play.sh" script

3. folder.conf before enabling resume
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cat ../shared/audiofolders/David\ Walliams\ -\ \(2016\)\ -\ Gangsta-Oma/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"

4 folder.conf after enabling resume via ssh (command used see point 2.)
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cat ../shared/audiofolders/David\ Walliams\ -\ \(2016\)\ -\ Gangsta-Oma/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

5. folder.conf after playing a different album
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cat ../shared/audiofolders/David\ Walliams\ -\ \(2016\)\ -\ Gangsta-Oma/folder.conf
CURRENTFILENAME="file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/David%20Walliams%20-%20%282016%29%20-%20Gangsta-Oma/01.%20Kapitel%201-%20Wasser%20mit%20Kohlgeschmack.mp3"
ELAPSED="23.474"
PLAYSTATUS="Stopped"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

6. folder.conf after swiping again the album that shall be resumed
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cat ../shared/audiofolders/David\ Walliams\ -\ \(2016\)\ -\ Gangsta-Oma/folder.conf
CURRENTFILENAME="file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/David%20Walliams%20-%20%282016%29%20-%20Gangsta-Oma/01.%20Kapitel%201-%20Wasser%20mit%20Kohlgeschmack.mp3"
ELAPSED="23.474"
PLAYSTATUS="Playing"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

Note: although it does say "playing" it does not play the album" after a second swipe it does play the album but from the start. The folder.conf does not change after a second swipe.

7. I changed the folder name to "1234" after reading issue #110

8. Now the resume function can be activated in the webinterface for this folder containing no whitespaces. Apparently resume still does not work, see below.

9. folder.conf of folder "1234" after activating resume in the webinterface
pi@raspberrypi:~/RPi-Jukebox-RFID/shared/audiofolders/1234 $ cat folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

10. folder.conf of folder "1234" after playing teh album and cahnging to a different one
pi@raspberrypi:~/RPi-Jukebox-RFID/shared/audiofolders/1234 $ cat folder.conf
CURRENTFILENAME="file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/1234/03.%20Kapitel%203-%20Klempner-Woche.mp3"
ELAPSED="8.797"
PLAYSTATUS="Stopped"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

11. folder.conf of folder "1234" after playing the album "1234" again. 
pi@raspberrypi:~/RPi-Jukebox-RFID/shared/audiofolders/1234 $ cat folder.conf
CURRENTFILENAME="file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/1234/03.%20Kapitel%203-%20Klempner-Woche.mp3"
ELAPSED="8.797"
PLAYSTATUS="Playing"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

Note: Although it says playing, it does not play the album. Swiping a second time or pressing play/pause starts the album from the beginning. Resume does not work.

# Analytics_AfterInstallScript.sh output #
Note that my version 1.1.7 did not have this script which is why I copy & pasted it from the github webinterface directly inte my hleperscripts directory
####################################
### PHONIEBOX INFO
### version
1.1.7
### edition
cat: /home/pi/RPi-Jukebox-RFID/settings/edition: No such file or directory

### Audio_iFace_Name
PCM

### Audio_Folders_Path
/home/pi/RPi-Jukebox-RFID/shared/audiofolders

### Audio_Volume_Change_Step
5

### Max_Volume_Limit
100

### Idle_Time_Before_Shutdown
0

### Second_Swipe
RESTART

### Playlists_Folders_Path
/home/pi/RPi-Jukebox-RFID/playlists

### ShowCover
cat: /home/pi/RPi-Jukebox-RFID/settings/ShowCover: No such file or directory

####################################
### CONF FILES DEFAULT

### /etc/samba/smb.conf
-rw-r--r-- 1 root root 9.5K Oct 27 14:44 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
### /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 1.1K Oct 22 00:27 /etc/lighttpd/lighttpd.conf
### /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 398 Oct 22 00:27 /etc/lighttpd/conf-available/15-fastcgi-php.conf
### /etc/php/7.0/fpm/php.ini
-rw-r--r-- 1 root root 70K Oct 22 00:27 /etc/php/7.0/fpm/php.ini
### /etc/sudoers
-r--r----- 1 root root 784 Oct 22 00:27 /etc/sudoers
### /etc/systemd/system/phoniebox*
ls: cannot access '/etc/systemd/system/phoniebox-rfid-reader.service': No such file or directory
ls: cannot access '/etc/systemd/system/phoniebox-startup-sound.service': No such file or directory
ls: cannot access '/etc/systemd/system/phoniebox-gpio-buttons.service': No such file or directory
ls: cannot access '/etc/systemd/system/phoniebox-idle-watchdog.service': No such file or directory
### /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Oct 26 00:04 /etc/mpd.conf
#music_directory                "/var/lib/mpd/music"
music_directory                "/home/pi/RPi-Jukebox-RFID/shared/audiofolders/"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
        mixer_control   "PCM"           # optional
#       mixer_control   "PCM"           # optional
####################################
### +Spotify Edition

### /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Oct 22 00:27 /etc/locale.gen
### /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Oct 22 00:27 /etc/locale.gen
### /etc/mopidy/mopidy.conf
-rw-r----- 1 mopidy root 791 Oct 22 00:28 /etc/mopidy/mopidy.conf
username =
### ~/.config/mopidy/mopidy.conf
-rw-r--r-- 1 root root 3.2K Oct 22 00:28 /home/pi/.config/mopidy/mopidy.conf
#username =
username =

# End of # Analytics_AfterInstallScript.sh output #

marcohorstmann, 2018-12-09:
I had last week a similar behavior. It seems there was an problem with the /settings/ folder. After moving all files to a new folder and restarted the box. Everything works. No idea yet why this helped but it works.

dtrv, 2018-12-11:
+1 Resume is not working

SimonChelkowski, 2018-12-11:
I have fixed the issue and plan to open a pull request as soon as I find the time to do so. 

Please be patient, help is on the way. Thanks.

MiczFlor, 2018-12-17:
Hi @SimonChelkowski thanks again for a detailed analysis. I need to call @princemaxwell into the discussion, because I have not spotify experience with Phoniebox. I am not sure how busy he is these days, because I haven't seen him for a while.

marvin78, 2019-02-20:
Is there any update on this topic.

Before I read this issue, I narrowed the exact same problem down to seek as @SimonChelkowski did. Is there a hotfix for this?

closed 2020-01-18 by MiczFlor