MiczFlor/RPi-Jukebox-RFID #353: Hifiberry Miniamp + Mopidy/alsasink issue

swyer222, 2018-11-27:
Hi all,

anyone else experiencing issues with Hifiberry devices? 

I'm having trouble with the Miniamp. Had this up and running with an earlier build, but stopped working as of recently. Tried setting the device up again with the latest RC, but no luck getting this running properly again. 

Issue: 
Bootup > OK
Boot sound > OK
speaker-test -D hifiberry -c 2 > OK until some time in the scripts, then noise just stops

Can't pinpoint the exact moment in the log, but think it stops sometime when Mopidy start or alsasink is set to aduio output. 

Appreciate any help pinpointing or fixing the issue. 

Thanks

cubertt, 2018-11-28:
Deaktivate the gpio script.

It was not enough to disable it via webapp, in my case i needed to run:

sudo systemctl disable phoniebox-gpio-buttons

closed 2018-11-28 by swyer222

internetfan420, 2018-12-07:
i had the same problem but i still need use the gpio pins because of the expander board i have and want to use buttons