MiczFlor/RPi-Jukebox-RFID #354: m3u and png files cause problems (solution exists and needs to be implemented)

SimonChelkowski, 2018-11-27:
I normally have in all my album folders an .m3u file which I use as playlist for the corresponding album on my usual software players and android devices. Sometimes album covers are present as .png files as well. This causes problems with Phoniebox. The problems are described in the following for three different scenarios:
1. album containing an m3u file (which is in my case always the last file as all track names start with the track number and the m3u filename corresponds to the album name which usually starts with a letter.)
2. album containing a .png file
3. album containing a m3u and a .png file

Error descriptions for scenario 1. only m3u: 
Music stops when m3u file is hit in playlist (either by ending previous song in playlist or by skipping until hitting the m3u file) but player does try to play something endlessly. It is somehow trapped in a loop. The webinterface shows constantly the "pause" symbol as if something is still played. Only pressing skip to next track (via gpio button or web interface) does bring the player to the end of the album and web interface shows "play" symbol to allow playing the album again (I repeat all is set, the album starts again as well).
While player is trapped in the loop and a different RFID card is swiped, the album is changed correctly but does not start playing. Even play/pause does not change that. Only "skip next" followed by "play/pause" causes the new album to start playing.

Error descriptions for scenario 2. only png: 
Music stops when png file is hit in playlist, other albums do not start (RFID triggered or via webinterface). Raspi needs to be rebooted!

Error descriptions for scenario 3. m3u + png: 
Same behaviour as m3u file only (as m3u is prior to the png file) when "skip next" is pushed, the png file is in the cue and the player wants to play it. The webinterface shows as track information no time, only 00:00. Other albums (RFID triggered or webinterface triggered) do not start. Raspi needs to be rebooted!

Solution (suggestion):
I developed a solution for the issue. As I am currently not familiar with github and modifying a branch to allow creating a pull request, I try to contribute in this way. 
The script "playlist_recursive_by_folder.php" needs to be modified.

The solution makes use of regular expressions which allows easy removal of certain file types (independent of the file names) from the array later forming the playlist of the Phoniebox for this album. 

Two modifications need to be done to the "playlist_recursive_by_folder.php" script

First modification: 
after the following line (right at the beginning for switching the debug mode)

	$debug = "false";

add the following function and comment.

	# added function to use regular expressions to remove multiple files e.g. all of the same file type from the array forming the later playlist. Idea originates from http://php.net/manual/en/function.array-diff.php#117219
	function array_preg_diff($a, $p) {
		foreach ($a as $i => $v)
			if (preg_match($p, $v))
				unset($a[$i]);
		return $a;
	}

Second modification:
Later in the script "playlist_recursive_by_folder.php"  one can find the following lines

		/*
            * ordinary, local files
            * list all files and folders
            * ignore . and ..
            */
            $folder_files = array_diff(scandir($folder), array('..', '.'));
			
I modified this section in the following way to remove *.m3u and *.png files from the &folder_files array

		/*
            * ordinary, local files
            * list all files and folders
            * ignore . and ..
            */
            # exchanged the following line by the next three as well as added the function array_preg_diff (see beginning of this file) to use regular expressions to remove multiple files e.g. all of the same file type from the array forming the later playlist. Idea originates from http://php.net/manual/en/function.array-diff.php#117219
		#$folder_files = array_diff(scandir($folder), array('..', '.'));
		$folder_files = array_preg_diff(scandir($folder), '/^\./'); # remove all files starting with "." meaning all files ".*" from array
		$folder_files = array_preg_diff($folder_files, '/^.*\.m3u$/i'); # remove all *.m3u files from array
		$folder_files = array_preg_diff($folder_files, '/^.*\.png$/i'); # remove all *.m3u files from array

I hope that this can be implemented directly and helps others as well.

MiczFlor, 2018-11-28:
Hi @SimonChelkowski 
awesome work and documentation. Thank you. I will look into this. And if you were to create a pull request, you could also add your name to the infamous list of code contributors. (or would you prefer me adding the enhancement)?
All the best, micz

SimonChelkowski, 2018-11-28:
Hi @MiczFlor 
Thanks for the kind words. I leave it to you to add the code directly. Reasons: 
1.  I am new to GitHub and not familiar with contributing as described by you in the [contributing section](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/CONTRIBUTING.md) 
2. I do not want do ruin my current install (Phoniebox for my kids which is in daily use) as I do not know what will happen when "I move to the `develop` branch" as you state in the getting started of the contributing section. I read through the [How To Create a Pull Request on GitHub](https://www.digitalocean.com/community/tutorials/how-to-create-a-pull-request-on-github) which you linked to, but It seems to me that I need to test this on a separate developer machine first to get familiar to it. 
Maybe you can comment on the following and thereby help? 
Do I need a second Phoniebox to start making contributions as my first one is in used with my kids (of course I can use it in the evening and night time)? Or in other words, can I have a working system based on my current installation and in parallel a develop branch on one Raspi?

The following describes what I have currently running: 
- I installed version 1.1.7 early in November 2018 and made several modifications which I could / would like to contribute.
- I make use of an RC522 reader and got it running ad described in [#11](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/11) 
- I use a 2x16 Matrix display (HD44780) to display date & time as well as track number and track title where the last one uses scrolling text. As this is something which is not part of your project (or at least not described, my highest interest is to contribute with this. The display code is running as a separate service. 
- I use five GPIO buttons (arcade style) to control `Vol+`, `Vol-`, `Prev`, `Next` and `Play/Pause` 
- I use a nice power button with a green LED that oscillates in brightness when the Raspi is shutting down
- other features are: separate USB sound card, internal 3A & 5V power supply,  4-channel relais module to turn of 5V and 3.3V devices (RFID reader, LEDs of arcade buttons, LCD display,...)

Thanks for your reply and suggestions. 
Best regards,
Simon

MiczFlor, 2018-12-07:
I merged this to the latest master:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/227fce04cb64fc991eefefeec89c202fe3cecd86
thanks to @SimonChelkowski @vansli

closed 2018-12-07 by MiczFlor

domu83, 2019-02-21:
Hi Simon,

I'm wondering if you would help me by providing your Display code? As I understood this thread this part of your above offering was not implemented yet, right?

Thanks in advance!!

best regards
domu

SimonChelkowski, 2019-02-22:
@domu83 
Please have a look at issue #498 which I just added. Everything you need to know is written there, I guess.

Cheers,
Simon