MiczFlor/RPi-Jukebox-RFID #359: No audio output after upgrading to 1.1.8-rc3

Alwiz79, 2018-12-05:
I'm a little embarassed since I had the problem before and cannot solve it now.
After upgrading to 1.1.8-rc3 + spotify edition, I don't have any sound output at all.
I have the hifiberry dac+ installed.

With 1.1.7 it worked fine. Can anyone give me a hint where to look at, as I am not the linux/raspberry specialist.?

I checked  /etc/asound.conf according to [https://support.hifiberry.com/hc/en-us/articles/205377202-Adding-software-volume-control](https://support.hifiberry.com/hc/en-us/articles/205377202-Adding-software-volume-control).

I checked /etc/mpd.conf according to my own thread [https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/195](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/195)

Alwiz79, 2018-12-05:
This has to be some config issue.
I can hear the bootup sound, but after that silence

Alwiz79, 2018-12-05:
Solution:
Deactivate the gpio script:
sudo systemctl disable phoniebox-gpio-buttons

Downside of this: GPIO pins cannot be used

XanomalieX, 2018-12-05:
Hmm, not a solution for me...

Alwiz79, 2018-12-05:
because it'S not working or because you lose the gpios?

XanomalieX, 2018-12-05:
Did not try, but no buttons are no option, I would stay with 1.7. Since I don't have a hifi berry, I'm not shure it's the same problem for my USB Soundcard.