MiczFlor/RPi-Jukebox-RFID #369: Sound funktioniert nicht

Ollimoller, 2018-12-07:
Hallo,

ich habe Phoniepbox 1.1.8 rc3 auf dem Raspberry Zero installiert.

Klicke ich im Webinterface auf Spotify sagt er die Seite kann nicht gefunden werden.
Zudem steht der Sound auf 0% und lässt sich nicht ändern

Die USB Soundkarte habe ich eingestellt und auch über alsamixer hochgeregelt.

Leider weiß ich nicht wodran dieses liegen könnte.

Gruß Michael

andisger, 2018-12-08:
Moin,
ich habe gestern die Phoniebox mit dem one line install command inklusive Spotify installiert.
es hat quasi alles direkt geklappt, nur hat er auch keinen Ton von Spotify gespielt und wenn ich im Web  Player auf Spotify geklickt habe, konnte er die Seite nicht öffnen.
Habe alles geprüft und quasi keinen Fehler gefunden.
Letztendlich habe ich festgestellt, dass mopidy Iris nicht installiert war. das habe ich manuell nachgeholt, neugestartet und dann lief es.

Allerdings hatte ich eine leicht andere Config: rasp 3 b+, sound noch von interner soundkarte

Ollimoller, 2018-12-10:
Danke, das ist es gewesen. Aus irgendeinen Grund schaffe ich es jetzt Songs, Playlists von Spotify abzurufen. Lokale Dateien laufen, Spotify leider nicht. Obwohl ich den Link direkt aus Iris kopiere.

Die Zugangsdaten scheinen ja auch alle zu passen. Ansonsten könnte ich ja nicht die Playlisten usw. suchen.

Gruß Michael

SimonChelkowski, 2018-12-13:
You can test if sound output works at all (independent of phoniebox) by running the command:

sudo aplay /usr/share/sounds/alsa/Front_Center.wav

If you hear above saying "front, center", then at least you know that the sound output works.

Ollimoller, 2018-12-15:
So, ich habe es jetzt zu ende getestet und festgestellt das es wohl am Pi-Zero liegt.

Auf einen Pi3 läuft alles ohne Probleme. Ich kann Spotify Playlisten abrufen usw.
Lege ich die SD-Karte allerdings in den Zero und ändere die Soundkarte laufen die örtlichen Dateien.
Beim abspielen von Spotify ruft er er wohl das Cover auf, startet allerdings die Playliste nicht. Bzw er hängt sich irgendwie auf. Die CPU-Auslastung ist immer bei 100%.

Ich denke einfach das der Zero nicht genügend Leistung hat. Also werde ich noch einen Pi3 bestellen.

Ein großes Dankeschön an alle die an diesem tollen Projekt mit arbeiten.