MiczFlor/RPi-Jukebox-RFID #370: turn raspi config to headphone jack not working

BigSlikTobi, 2018-12-08:
Hi guys, 

at the moment I am trying to start my phoniebox and test the first sounds. 
Unfortunately it keeps quiet at the moment. I tried different ways to get sound. Wireless, with a USB - Soundcard and directly. My box doesn´t even react if I start a song it just reloads and does nothing. 
I searched around a little bit and it turns out that I shall turn raspi config to headphone jack to create sound.
But as I was trying this I got the error message that "headphone jack is not working". The same error appears when trying to select hdmi sound. 
Does this issue sounds familiar to anyone and do you have a solution for me?

Thanks!
Tobias

tobiasnix, 2018-12-17:
Had a similar issue a while back. Cant remember the solution but its working now.

Try to get some native sound:
```
aplay --device=hw:0,0 /usr/share/sounds/alsa/Front_Center.wav  (audio jack on my setup)
aplay --device=hw:1,0 /usr/share/sounds/alsa/Front_Center.wav  (usb soundcard on my setup)
```

If that does work, check the audio configuration files:
```
pi@raspberrypi:~ $ sudo vi /usr/share/alsa/alsa.conf (defaults.ctl.card 0 && defaults.pcm.card 0)
pi@raspberrypi:~ $ sudo vi /etc/mopidy/mopidy.conf (volume, ...)
```
Maybe some of this will help you.

MiczFlor, 2018-12-17:
Hi @BigSlikTobi 
Does the Pi play audio when you are booting with HDMI into Raspian, i.e. the desktop environment?
If and when you solve it, please add this under troubleshooting in the wiki or post it here, so I can add it. Thanks!

BigSlikTobi, 2018-12-18:
Hi Tobias & MiczFlor,

I didn‘t had the chance to try your aproaches yet but it turned out that I have to restart mpd every time after uploading new folders. I don‘t really understand why uploading a new folder with a mp3 file turns the sound output into error but I‘ll try out your tipps as well as some other aproaches I googled after xmas and keep you updated.

Merry Xmas! 
Tobias

MiczFlor, 2018-12-26:
Hi @BigSlikTobi 
nutzt du die +Spotify Edition oder "Classic"? Für Spotify gibt es dazu wichtige Anmerkungen (die ich jetzt auch mal ins WIKI schreiben werde):
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/SPOTIFY-INTEGRATION.md

MP3 (local music) handling has completely changed for the +Spotify Edition. The tracks need to be indexed by Mopidy & Mopidy-Spotify (this is the part of Phoniebox, which gives you spotify support) and the created M3U files do have another structure than normal M3U files.
* You have to scan you library once (if you hadn't yet) and everytime you upload new local tracks to your Phoniebox.
* You can scan local files under "Folders & Files" (Top Navigation).
* For the future we try to integrate an automation for that.

(the last item is surely still on the todo list :)

closed 2019-06-21 by MiczFlor