MiczFlor/RPi-Jukebox-RFID #372: Why not to start with the PiMuxicBox Image?

mr-gosh, 2018-12-09:
Wouldn't it be easier to start with a http://www.pimusicbox.com/ Image instead of a plain debian?

mr-gosh, 2018-12-11:
ok - makes sense - but the image boots much faster...
I started again with a mini installation and there we go!
Boot time is now much lower...

cubertt, 2018-12-21:
> ok - makes sense - but the image boots much faster...
> I started again with a mini installation and there we go!
> Boot time is now much lower...

How much?
I get 12 sec with the recommended installation until the boot up sound is ... but i would be interested tio reduce the time even more (Raspi 3 a+, MFRC522, i2s Dac).

Does phoniebox install (via one click install) without problems?

mr-gosh, 2018-12-26:
OK - my boottimes are much slower.
I have a rpi2 
It takes 12 seconds until the rfid beeps and another 30 seconds until boot up sound.

With gui it was twice the time...
I have a rpi3b+ lying around - can i boot the same image or do i have tonsetbit up individually?
I could test boot time there.

Install scripts ran through without problems i think.

closed 2019-06-21 by MiczFlor