MiczFlor/RPi-Jukebox-RFID #375: GPIO Buttons reagieren nicht

alienfactor, 2018-12-10:
Moin,

ich hab die Buttons entsprechend der Anleitung angeschlossen (ohne Pullup-Widerstände)
Das gpio-buttons.py Script habe ich 1:1 übernommen.

Die Buttons funktionieren jedoch nur sporadisch. In der UI sehe, dass die Button ständig wieder deaktiviert werden. Ein manuelles aktivieren bringt aber auch nicht mäßigen und maximal temporären Erfolg.

Auch die in #62 genannte Anpassung funktioniert nicht (aus dem Bild entnehme ich, dass 5.3k verwendet wurden, ich habe es mit 4.7k versucht).

Jemand eine Idee warum die GPIOs so überhaupt nicht wollen?
Vielen Dank

alienfactor, 2018-12-10:
.. fehlerhafte gpio-buttons.py verwendet. Nach Korrektur funktioniert es.

closed 2018-12-10 by alienfactor

MiczFlor, 2018-12-10:
>.. fehlerhafte gpio-buttons.py verwendet. Nach Korrektur funktioniert es.

@alienfactor kannst du noch mal ganz kurz hier schreiben was fehlerhaft war? Dann würde ich das a) korrigieren und b) im manual dokumentieren. Für die nächsten sozusagen. Danke und schlaf dich aus (hoffe ich) :)
Beste Grüße, Micz 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

alienfactor, 2018-12-10:
@MiczFlor Hallo und danke für deine tolle Missionsarbeit hier! :)
Der Fehler ist durch mein eigenes Verschulden aufgetreten. Ich bastelte an der Datei herum um #62 umzusetzen, danach ging nix mehr. Ich habe also mit "Korrektur" nur den Ausgangszustand wiederhergestellt. Sorry für die unklare Formulierung.

Vielleicht darf ich aber dein Projekt-Wissen dennoch kurz in Anspruch nehmen und etwas zu #62 fragen:
Ich habe den Messpunkt (B) nach außen geführt und mit einem Widerstandund mit GPIO6 verbunden. Tut sich leider gar nix. In der Anleitung steht etwas von "the Pin has to be pulled down".. Vermutlich muss ich noch irgendwo gegen GND schalten? Hast du eine Erklärung?

Vielen Dank!