MiczFlor/RPi-Jukebox-RFID #378: httpd error.log grows with php errors

dtrv, 2018-12-11:
`/var/log/lighttpd/error.log` grows with lots of php warnings and errors. This is bad for the SD-card and it cost performance (especially on my RPi zero). I got > 5M log file just from installing and playing some minutes with the web gui. Here are the top 15:

```
root@raspberrypi:~# cat /var/log/lighttpd/error.log | cut -d" " -f 4- | sort | uniq -c | sort -nr | head -n 15
   2823 FastCGI-stderr: PHP message: PHP Notice:  Undefined variable: lang in /var/www/jukebox/htdocs/ajax.loadControls.php on line 30
   2538 FastCGI-stderr: PHP Notice:  Undefined variable: lang in /var/www/jukebox/htdocs/ajax.loadControls.php on line 30
   1707 FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 41
   1707 FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 38
   1707 FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 35
   1707 FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 32
   1707 FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 29
   1695 FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 26
   1695 FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 23
   1151 FastCGI-stderr: PHP message: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/ajax.refresh_id.php on line 8
    943 FastCGI-stderr: PHP message: PHP Notice:  Undefined variable: lang in /var/www/jukebox/htdocs/ajax.loadControls.php on line 24
    846 FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Played_File): failed to open stream: No such file or directory in /var/www/jukebox/htdocs/ajax.loadCover.php on line 8
    846 FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /var/www/jukebox/htdocs/ajax.loadCover.php on line 7
    846 FastCGI-stderr: PHP Notice:  Undefined variable: lang in /var/www/jukebox/htdocs/ajax.loadControls.php on line 24
    823 FastCGI-stderr: PHP message: PHP Notice:  Undefined offset: 1 in /var/www/jukebox/htdocs/inc.playerStatus.php on line 41
```

I eliminated some errors by simply checking variable state in some scripts. But I never coded PHP before and don't want to start ...

Note that my code is located in `/var/www` and it's running by user `www-data` - the same the httpd is running. I always had trouble with rights and ownerships of the files and don't see why another user (`pi`) should be involved. Giving all users sudo power should never be a solution IMO.

Are only my logs full of errors? What am I missing?

RogoTe, 2019-07-06:
No, it is not only you. As long as code is not cleaned up, it is a workaround to switch PHP warnings off:

     cd /etc/php/7.0/fpm/conf.d/
     sudo touch 999-RPi-Jukebox-RFID.ini
     sudo sh -c 'echo "error_reporting = E_ALL & ~E_NOTICE & ~E_STRICT & ~E_DEPRECATED & ~E_WARNING;" >> 999-RPi-Jukebox-RFID.ini'
     sudo service php7.0-fpm restart

MiczFlor, 2019-07-07:
Hi @RogoTe 
I am sure you know how long it takes to clean up code :) which is why I would like you to put this hack into the pull request which you offered for the caching in another thread. Thanks and: good work!

closed 2020-01-18 by MiczFlor