MiczFlor/RPi-Jukebox-RFID #384: Please do not automatically activate GPIO Buttons - breaks Phat Beat DAC

ShaXiaomi, 2018-12-12:
Hello,

I am using the Phat Beat DAC with Phoniebox.
The sample GPIO config which is copied by the installer breaks to Phat Beat DAC.
See Phat Beat GPIO config: https://de.pinout.xyz/pinout/phat_beat

It would be more preferable to make the GPIO stuff optional.

thx
Daniel

dtrv, 2018-12-17:
Good to know. I plan to use PhatBeat with my next Box. Thanks for reporting ...

MiczFlor, 2018-12-18:
Hi @ShaXiaomi @dtrv could you add this to the wiki? Im troubleshooting? Or under soundcards?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

martinclausen2, 2018-12-28:
Hi @ShaXiaomi , could you please add the Phat Beat DAC pinout in the [wiki?](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#hardware-with-fixed-pin-assignment). This will help the community in the future to prevent conflicting GPIO assignments.

ShaXiaomi, 2018-12-28:
Hi, I already added something here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ

But not yet the pin out. Will do it when I am not on mobile.

closed 2020-01-18 by MiczFlor