MiczFlor/RPi-Jukebox-RFID #385: One line installation errors of version 1.1.8

SimonChelkowski, 2018-12-12:
**Introduction**
This week I was setting up a test system (Pi 1 Model B Revision 2.0 with two SD cards, one for classic and one for the Spotify version) besides my running Kids version (Pi 3 B+, Version 1.1.7+Spotify) and encountered two errors during the installation of both versions (classic and Spotify version). I will list the errors in the following before giving any details and suggestions on how to solve it. Note that I used the 
lite version of stretch for the installation of stretch with filename: _2018-11-13-raspbian-stretch-lite.zip_.

**Naming the two errors**
1. pi-rc522 produced an import error (classic & spotify version)
2. error during mopidy installation, ".config/mopidy" folder cannot be created (spotify version only)

**Details of the errors and suggestions for solutions:**

1. 
The console output during installation when producing the error looks like the following
```
Collecting pi-rc522
  Downloading https://files.pythonhosted.org/packages/84/f2/e3d02257949e9caa9bff                                   26044e0185e743ff45d3e0e099a93880e10bf718/pi-rc522-2.2.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-UZqi_X/pi-rc522/setup.py", line 11, in <module>
        from pirc522 import __version__  # flake8: noqa
      File "pirc522/__init__.py", line 4, in <module>
        from .rfid import RFID
      File "pirc522/rfid.py", line 2, in <module>
        import spidev
    ImportError: No module named spidev

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-UZqi_X/pi-rc522/
```
This error I could correct by installing spidev running the command
`pip install spidev`
and afterwards installing pi-rc522 again by running the command
`pip install pi-rc522` 

2. 
The console output during installation when producing the error looks like the following
```
Synchronizing state of mopidy.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable mopidy
Generating locales (this might take a while)...
  en_GB.ISO-8859-1... done
  en_GB.UTF-8... done
Generation complete.
mkdir: cannot create directory ‘/home/pi/.config/mopidy’: No such file or directory
cp: cannot create regular file '/home/pi/.config/mopidy/mopidy.conf': No such file or directory
sed: can't read /home/pi/.config/mopidy/mopidy.conf: No such file or directory
sed: can't read /home/pi/.config/mopidy/mopidy.conf: No such file or directory
sed: can't read /home/pi/.config/mopidy/mopidy.conf: No such file or directory
sed: can't read /home/pi/.config/mopidy/mopidy.conf: No such file or directory
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dhcpcd
...
```
This error I could correct by adapting the command of the install script. The original command producing the error is
`sudo mkdir /home/pi/.config/mopidy`
It seems that nested new folders (new folder in a new folder) cannot be created by mkdir, which is why I separated the commands into 
```
sudo mkdir /home/pi/.config/
sudo mkdir /home/pi/.config/mopidy
```

**Final words**
I wonder if anybody else experienced these problems? But I assume that these bugs are so general that they have nothing to do with my hardware. The first one might be related to the fact that I used the lite version of stretch for the installation.

tobiasnix, 2018-12-17:
I can confirm #1 error (classic). And also that the solution (pip install spidev; pip install pi-rc522) seems to fix it.

MiczFlor, 2018-12-18:
Hi @SimonChelkowski @tobiasnix 
Does this relate to all installs or a specific hardware?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

SimonChelkowski, 2018-12-18:
Hi @MiczFlor 
I experienced this with both 1.1.8 versions (classic and spotify). I don't believe that it is hardware dependent, it is to fundamental. @tobiasnix what Raspi version did you use when you experience the same error?

tobiasnix, 2018-12-18:
Its classic on my side with raspberry 2 and 3 B

commit c28b6754 (2018-12-23): 1.1.9-rc1 bug fixes #385
closed 2018-12-25 by MiczFlor