MiczFlor/RPi-Jukebox-RFID #386: (ver. 1.1.8) RC522 module does not detect cards, but pi-rc522 example does work and shows card-IDs)

SimonChelkowski, 2018-12-12:
This week I installed a test system (Pi 1 Model B Revision 2.0 with two SD cards, one for classic and one for the Spotify version). The setup included a RFID-RC522 card reader. All lines of the RFIID module including IRQ connection are connected as described on [pi-rc522 github page](https://github.com/ondryaso/pi-rc522). Nevertheless in neither of the two 1.1.8 phoniebox version I installed the command
`python ~/RPi-Jukebox-RFID/scripts/RegisterDevice.py`
results in the module being listed. Hence, I cannot select any. 

I used the [pi-rc522 Examples](https://github.com/ondryaso/pi-rc522/tree/master/examples) called _Read.py_ to test the RFID reader. The result is shown in the following

```
Starting

Detected: 10
Card read UID: 252,3,178,121
Setting tag
Selecting UID [252, 3, 178, 121, 52]

Authorizing
Changing used auth key to [116, 0, 82, 53, 0, 255] using method B

Reading
Calling card_auth on UID [252, 3, 178, 121, 52]
Error on S1B0

Deauthorizing
Changing auth key and method to None
```
As you can see the RFID module does work properly and cards are detected and read out, at least by this example. Nevertheless the _RegisterDevice.py_ script and the webserver do not show any Card-ID (even if I create the file _deviceName.txt_ manually with the content 
`MFRC522`
manually.

Any suggestion on how to make it work?

alex-voigt, 2018-12-15:
I had the same problem and fixed it in this pull request #374.
Maybe you can test and confirm that it works.

Copy the files of the pull request over to your script folder. Then you should be able to register the device again. Copy over experimental reader over the normal reader and everything should work again.

SimonChelkowski, 2018-12-16:
Great job, it works like a charm now. My MFRC522 card reader is now registered and CardIDs show up in the web interface. I used the files from the current develop branch as your pull request has already been added to it some hours ago. 

Thanks!

MiczFlor, 2018-12-18:
Hi @SimonChelkowski @alex-voigt 
has anyone tested this with other card readers including USB readers?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

SimonChelkowski, 2018-12-18:
Sorry I cannot test it with a USB card reader as I only have non-USB card readers.

alex-voigt, 2018-12-18:
@MiczFlor 
I tested the experimental reader with a Neuftech USB reader and it worked. As I have written in the pull request the normal reader doesn't work right now without applying a fix.

Ardios, 2018-12-18:
@MiczFlor @alex-voigt ,

work for me, copy 3 Files from the developer branch (RegisterDevice.py, Reader.py.experimental renamed to Reader.py and daemon_rfid_reader.py) then run `python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py` and Done...thanks :)

MiczFlor, 2018-12-25:
merging new code referencing this thread in a minute to master.

closed 2018-12-25 by MiczFlor