MiczFlor/RPi-Jukebox-RFID #390: WLAN Password visible in WebUI

dMopp, 2018-12-17:
The WLAN Password is CLEARTEXT visible in the NOn protected WebUI, this is a big security issue  :(

dMopp, 2018-12-17:
My small hotfix fow now:

```
Add the following to "/etc/lighttpd/lighttpd.conf"
auth.backend                = "plain"
auth.backend.plain.userfile = "/home/pi/RPi-Jukebox-RFID/password.txt"

auth.require = (
"/settings.php" =>
   (
      "method" => "basic",
      "realm" => "Password protected area",
      "require" => "user=admin"
   )
)
```

And create a file called "/home/pi/RPi-Jukebox-RFID/password.txt" with following content:
`admin:secretpassWord`

Maybe it makes sende to ask for password protection during the installer and let him create files and settings...

tobiasnix, 2018-12-17:
Or just configure the wifi via wpa_supplicant.conf during base system setup. Thats how i do it and nothing is displayed in the web form.

dMopp, 2018-12-17:
Thats what i did. Its reading it out of the file and showing it cleartext... (Raspian Stretch)

tobiasnix, 2018-12-17:
Had a look into https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/inc.setWifi.php and it seems like it should show the password. In my case its not displayed because the file permission is too restrictive:

-rw------- 1 root root 153 Dec 17 09:01 /etc/wpa_supplicant/wpa_supplicant.conf

(in my case its copied over during base install from /boot/wpa_supplicant.conf)

dMopp, 2018-12-17:
Makes sense, so the questions are: 
- Are the p[ermissions wrong?
--> if yes: Why? (installer? WebUI?)
--> if no: Maybe change the behavior of webUI OR add the passwd part.

dMopp, 2018-12-17:
> Hi @dMopp @tobiasnix
> how much of a security issue is this really?
> I added the wifi password form while working on the autodetect hotspot. My use case was:
>
> Phoniebox at home: works on the local wifi
> Take Phoniebox to a holiday location with a wifi
> Now you need an easy way to add ssid and password to the Phoniebox
> Auto hotspot will turn the Phoniebox into a hotspot that you can connect to from your phone
> Now you could use the Phoniebox via the phone, without internet
> Using the Web UI you can now change ssid and password and you are set up like "home away from home"
> That's why I have it there and why it is a simple plain text form. And, of course, the hotspot on demand never got finished :)

The Problem:

Now it’s showing the Data used by wpa_supplicant to connect to my WiFi at home.

MiczFlor, 2018-12-17:
Hi @dMopp 
it is showing the wifi password to everyone in the wifi network, i.e.: everyone who has the password. Or am I missing something?

dMopp, 2018-12-17:
In Theorey yes, but normally my kids didn’t know the password. They have devices logged in or (plan for the future) using profiles containing the informations without knowing the real password. Anyway it might make senses to password protect the whole UI or the settings page.. 

Mit freundlichen Grüßen
Daniel Quolke

MiczFlor, 2018-12-17:
Hi Daniel,
I see :) would this be enough protection?
https://developer.mozilla.org/en-US/docs/Web/HTML/Element/input/password

Ardios, 2018-12-17:
Hey,

ich antworte dir einfach mal ganz frech auf Deutsch, mein Basis Englisch ist leider echt schlecht :(

Also wenn es dir nur darum geht, das bei dir persönlich die SSID und das Kennwort nicht angezeigt werden sollen, so kann ich dir nur den weg von @tobiasnix empfehlen...

1.) Raspbian auf deine SD Karte schreiben
2.) Direkt am PC die boot Partition von der SD Karte öffnen
3.) wpa_supplicant.conf am PC mit folgendem Inhalt anlegen:
`# Datei wpa_supplicant.conf in der Boot-Partition (Raspbian Stretch)
country=DE
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
       ssid="WLAN SSID"
       psk="DEINPASSWORT"
       key_mgmt=WPA-PSK
}`
4.) Die wpa_supplicant.conf direkt ins Hauptverzeichnis auf der Boot Partition kopieren
5.) Phoniebox Installation starten und bei der Installation die Wifi Einstellungen mit Nein (N) überspringen

Wenn du es auf diesem Weg machst, wird dir in der Phoniebox WebGUI weder SSID noch Passwort angezeigt.

Bei deinem bestehenden System sollte es eigentlich ausreichen die Rechte der wpa_supplicant.conf anzupassen.

Grüße

MiczFlor, 2018-12-23:
I added this little fix to the `develop` branch. Will be merry merged after merry christmas

closed 2018-12-23 by MiczFlor