MiczFlor/RPi-Jukebox-RFID #391: Play files from Subfolders with assigned card ID

tobiasnix, 2018-12-17:
I have organized my files in the following way:
```
/Folder A
/Folder A/Subfolder A/File A
/Folder A/Subfolder B/File B
/Folder A/Subfolder C/File C
```
Each subfolder (eg. `Subfolder A`, `Subfolder B`, ...) got its own card ID and its playing the correct file.
`Folder A` also got a card ID but does not play File A to C.
Is that possible and how can i got it working?
Playing is possible if i start it through webui.

```
$ sudo /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh 
************************************
*** PHONIEBOX INFO
*** version: 1.1.8
*** edition: classic
*** Audio_iFace_Name: PCM
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 10
*** Second_Swipe: RESTART
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 9.5K Dec 17 11:39 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 1.1K Dec 17 11:40 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 398 Dec 17 11:40 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
-rw-r--r-- 1 root root 70K Dec 17 11:40 /etc/php/7.0/fpm/php.ini
*** /etc/sudoers
-r--r----- 1 root root 784 Dec 17 11:40 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 313 Dec 17 11:40 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Dec 17 11:40 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 272 Dec 17 11:40 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Dec 17 11:40 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Dec 17 11:40 /etc/mpd.conf
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
	mixer_control	"PCM"		# optional
#	mixer_control	"PCM"		# optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 1.7K Sep 10 09:31 /etc/dhcpcd.conf
#static ip_address=192.168.0.10/24
#static ip_address=192.168.1.23/24
#static routers=192.168.0.1
#static routers=192.168.1.1
option domain_name_servers, domain_name, domain_search, host_name
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
#static domain_name_servers=192.168.1.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw------- 1 root root 153 Dec 17 09:01 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition
 
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Nov 13 13:07 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Nov 13 13:07 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: cannot access '/etc/mopidy/mopidy.conf': No such file or directory
cat: /etc/mopidy/mopidy.conf: No such file or directory
*** ~/.config/mopidy/mopidy.conf
ls: cannot access '/root/.config/mopidy/mopidy.conf': No such file or directory
cat: /root/.config/mopidy/mopidy.conf: No such file or directory
```

tobiasnix, 2018-12-20:
Tried to gather some more informations: All files and folders are renamed (no whitespaces or special characters) and all permissions are set to 777/666

```
pi@raspberrypi:~/RPi-Jukebox-RFID/shared/audiofolders $ ls -alR Folder-A
Folder-A:
total 20
drwxrwxrwx 5 pi  www-data 4096 Dec 20 15:34 .
drwxrwxrwx 7 777 pi       4096 Dec 20 15:19 ..
drwxrwxrwx 2 pi  www-data 4096 Dec 20 15:30 Subfolder-A
drwxrwxrwx 2 pi  www-data 4096 Dec 20 15:30 Subfolder-B
drwxrwxrwx 2 pi  www-data 4096 Dec 20 15:33 Subfolder-C

Folder-A/Subfolder-A:
total 8712
drwxrwxrwx 2 pi www-data    4096 Dec 20 15:30 .
drwxrwxrwx 5 pi www-data    4096 Dec 20 15:34 ..
-rwxrwxrwx 1 pi www-data 8909833 Dec 20 15:19 1.flac

Folder-A/Subfolder-B:
total 8164
drwxrwxrwx 2 pi www-data    4096 Dec 20 15:30 .
drwxrwxrwx 5 pi www-data    4096 Dec 20 15:34 ..
-rwxrwxrwx 1 pi www-data 8349056 Dec 20 15:19 2.mp3

Folder-A/Subfolder-C:
total 4472
drwxrwxrwx 2 pi www-data    4096 Dec 20 15:33 .
drwxrwxrwx 5 pi www-data    4096 Dec 20 15:34 ..
-rwxrwxrwx 1 pi www-data 4567653 Dec 20 15:19 3.mp3
```

I have cleared playlist folder and tried to start the recursive folder play with the card (not working) and also through the webinterface (working).

```
pi@raspberrypi:~/RPi-Jukebox-RFID/playlists $ ls -la
total 12
drwxrwxrwx  2 pi   pi   4096 Dec 20 15:37 .
drwxr-xr-x 11 pi   pi   4096 Dec 20 14:17 ..
-rw-r--r--  1 pi   pi      0 Dec 20 15:37 Folder-A.m3u
-rw-r--r--  1 root root  220 Dec 20 15:37 Folder-A %RCRSV%.m3u
```
The playlist generated by the card-id is empty (Folder-A.m3u) the playlist generated through the webinterface does list the files. 
Seems like the problem is related to an empty generated playlist as using the card id vs a full playlist playing from the webinterface.

tobiasnix, 2018-12-20:
Found the file https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/rfid_trigger_play.sh and the option `-v=recursive`. Its working with that option attached but only in shell and through webinterface. Seems like this option has to be passed to the script. How can i add this parameter to a card id? Am i missing a configuration option or an option in the webinterface?

MiczFlor, 2018-12-23:
HI @tobiasnix 
sorry for the late reply, there is no way to add a card for recursive subfolder play at the moment. A card linked to a folder at the moment will only play the files inside the folder itself (and should also play webradio and the like) but not dive into sub-directories.
If you have an idea on how to add this, please share it via pull request and/or the issue threads. Don't forget: the Web UI would also need an option to add either "play content only" or "play subdirs, too".

closed 2020-01-18 by MiczFlor

driebe, 2024-03-08:
Hi @MiczFlor 

coming back to this feature request, would it make sense to model this via the "playlist" functionality already present? So for instance, one could already add all relevant files via the existing ui to the current playlist. The two only things which would be missing are 

1) a way to persist the playlist by saving it like with other media players
2) a new hook for the card reader where you could assign a saved playlist to the cards (instead of folders)

Sorry, I am not a software engineer, so I can't contribute this in form of a pull request.

Regards
Daniel

s-martin, 2024-03-08:
Hi @driebe,
could you open a new feature request?

Comments in already closed issues get overlooked easily.